adangert/JoustMania #3: Unable to complete setup.sh

Trikkitt, 2016-08-25:
I've followed the instructions using the latest version of Raspian.  The following errors occur while running setup.sh

CMake Error: Error in cmake code at
/home/pi/psmoveapi/CMakeLists.txt:3:
Parse error.  Expected a command name, got unquoted argument with text "<<<<<<<".
-- Configuring incomplete, errors occurred!
make: *** No targets specified and no makefile found.  Stop.
sed: can't read /etc/modprobe.d/alsa-base.conf: No such file or directory

Broadcast message from pi@raspberrypi on pts/0 (Thu 2016-08-25 20:08:04 UTC):

The system is going down for reboot NOW!

aangert, 2016-09-08:
Hey there, I've tested this with wheezy only, I had some problems with Jessie, I might go back and work on them, but for now you could try using the image here and see if it works: https://downloads.raspberrypi.org/raspbian/images/raspbian-2015-05-07/

Parse error. Expected a command name, got unquoted argument with text "<<<<<<<".
this somewhat looks like a git merge conflict

sed: can't read /etc/modprobe.d/alsa-base.conf: No such file or directory
and this looks like Jessie changing up how the sound system works, alsa-base.conf is apart of raspians sound system.

Trikkitt, 2016-09-09:
Thanks - I've successfully run everything, but it isn't pairing any controller.  There was an error around the PSMoveAPI complaining about -lrt switch, which I've seen elsewhere with someone trying to get it to work with the Raspberry Pi.  I'm guessing that it is another change that has broken something since you got all this working!

Thanks for the pointers so far!

aangert, 2016-09-10:
yeah no problem, when you start system, you should just be able to plug in a controller via usb to the raspberry pi and it should then connect. The logic for how it connects is in the pair.py file

Trikkitt, 2016-09-10:
I've managed to get it up and running now.  Ended up manually downloading the PSMoveAPI and compiling it, and also located the new path for the alsa-base.conf, and manually installing a lot of pre-reqs.  But I got there in the end!

Biggest issue that I'm battling is being unable to run two bluetooth dongles on the same RPi.  Just one dongle works, supports 7 devices and works reliably.  Add a second one and the bluetooth daemon stops the moment the move controllers start connecting.  Reboot with just one or the other and it works fine, run it with two and it fails, I've used two different models, two of the same, and the same issue.  Sorry for the issue spread!

aangert, 2016-09-11:
you can check all of the connected BT connectors and devices in '/var/lib/bluetooth/' to help troubleshoot the problems you might be having, each folder in that directory should be a dongle's mac address, and the Bluetooth controllers should be spread evenly among the different dongles, that's what the pair.py does again, it might be something new with Jessie, also if you are using a Rasberry pi 3, it has built in bluetooth, which might need to be disabled first, and should show up in the /var/lib/bluetooth path.

and again, I haven't used Jessie, so there might be a couple more issues there too. I'll see about updating the repo to take care of that when I get around to it, but if you also wanted to put in a pull request that would be appreciated :)

aangert, 2016-09-11:
just pushed up some changes that included jessie support, you might want to give it a go again.

adangert, 2016-09-12:
oh hey, @Trikkitt can I ask how you build the psmove repo? I was going through a clean installation and ran into a few problems.

adangert, 2016-09-12:
Hey @Trikkitt I've run through the Jessie install from scratch and added some things that were causing it to break, including the bluetooth daemon shutting off, should work with multiple dongles, although something is wrong with the audio/music still, so I'll have to look more into that.

adangert, 2016-10-01:
heya! everything should now be working from scratch including sound with the latest raspberry pi Jessie build

closed 2016-10-01 by adangert