adangert/JoustMania #11: Random teams start

adangert, 2016-12-11:
Allow random teams to congregate before starting the game

etaiklein, 2017-01-11:
the main problem here seems to be that the division of teams before and after the game starts are different.

i.e. 
State 1 (On team mode selected): the controllers are divided into teams
State 2 (Play Begins): teams are divided again.

--------

another possible solution would be:
State 1: controllers flash between team colors (blue/red)
State 2: teams solidify (blue or red)

adangert, 2017-01-12:
well before the game starts there are no assigned teams for random mode, what a solution could be is once everyone has joined in (by pressing A) the teams are assigned and then after about ten seconds the game starts, which I think you said above. The problem in the game right now is that the timer counts down and then the teams are immediately assigned.

adangert, 2017-02-25:
the game will pause before the countdown for six seconds allowing team members to group up before the game starts

closed 2017-02-25 by adangert