adangert/JoustMania #8: Bullets are unbalanced in Zombie mode

jouva, 2016-12-11:
At the start of the game, it's very easy to have a bullet, and with enough coordination, zombies could be held off purely with bullets. It seems like something that should be delayed before giving them out, or spawn longer or such

adangert, 2016-12-11:
good point, maybe, there should be a multiplier for how many zombies there are

jouva, 2016-12-11:
Yeah I brought this up a week ago at a convention. There's multiple ways it could be handled, all of which will need tweaking and play testing:

- Timer before bullets are even available
- Number of kills before bullets are available to a player
- Maths involving number of zombies (like you mentioned)

Basically some combination of player count and time and other factors involved in calculating when they should get one

adangert, 2017-01-11:
quick fix: bullets are only assigned after 4 zombies have died

closed 2017-01-11 by adangert