adangert/JoustMania #24: random crashes when starting some games

adangert, 2017-01-11:
sometimes when starting a game the pi crashes,
perhaps try putting a time.sleep in between new controller processes in each game?

adangert, 2017-01-14:
looks like it might be because of  CPU spikes when python makes a bunch of processes, perhaps limiting the processes would help, the game usually shuts off within the first 10 minutes of starting the game, and looks like it crashes less when there is no screen attached.

adangert, 2017-06-05:
also could be related to heat/memory issues the pi is experiencing under stress, investigation is needed.

adangert, 2018-01-08:
Looks like it is due to a memory allocation error.

adangert, 2018-01-08:
There could be a memory leak in the code somewhere, investigation would be necessary

adangert, 2018-03-08:
@chaintip

chaintip, 2018-03-08:
`0.00518358 BCH`| `~   6.73 USD` has [been tipped](https://explorer.bitcoin.com/bch/address/bitcoincash:qqh0ld4rxyvqxlcpcc9anfdvh8sg744kqysflydp4k) to this issue by @adangert.
To claim [this bounty](http://github.chaintip.org/chaintip/bounties/issues/22), get a pull request merged with ```@chaintip fixes #24``` in the creation comment.

chaintip, 2018-03-08:
The fiat value for this is incorrect. Will update soon.

block-spider, 2018-03-08:
@chaintip 2 days

chaintip, 2018-03-08:
Thank you @block-spider! `0.00994548 BCH`| `~  10.45 USD` has now been tipped to this issue.
To claim [this bounty](http://github.chaintip.org/chaintip/bounties/issues/22), get a pull request merged with ```@chaintip fixes #24``` in the creation comment.

chaintip, 2018-03-10:
`0.00518358 BCH`| `~   5.53 USD` has now been tipped to this issue.
To claim [this bounty](http://github.chaintip.org/chaintip/bounties/issues/22), get a pull request merged with ```@chaintip fixes #24``` in the creation comment.

derianalanrojas, 2018-04-01:
The stress of the processing could also be due to the type of CPU processor often influences too much

chaintip, 2018-04-07:
`0 BCH`| `~   0.00 USD` has now been tipped to this issue.
To claim [this bounty](http://github.chaintip.org/chaintip/bounties/issues/22), get a pull request merged with ```@chaintip fixes #24``` in the creation comment.

adangert, 2019-01-14:
Fixed with stability update

closed 2019-01-14 by adangert