adangert/JoustMania #28: Issue with swig_module_helper

afriendcalledben, 2017-01-16:
Hi, 

I've been wanting a portable version of Joust since I first saw years ago so this is awesome work you're doing.

I'm trying to set it up at the moment but am having problems with the psmove module. Initially it wasn't finding it all so I added:
`import sys
sys.path.append('/home/pi/psmoveapi/build/`

Now I'm getting a:
`Traceback (most recent call last):
  File "pair.py", line 3, in <module>
    import psmove
  File "/home/pi/psmoveapi/build/psmove.py", line 28, in <module>
    _psmove = swig_import_helper()
  File "/home/pi/psmoveapi/build/psmove.py", line 24, in swig_import_helper
    _mod = imp.load_module('_psmove', fp, pathname, description)
ImportError: dynamic module does not define init function (init_psmove)
`

I'm working with a fresh install of Raspbian on a Raspberry Pi 3 v1.2. I've got SWIG version 2.0.12. Any tips as to where I might be going wrong would be really appreciated.

Thanks,

Ben

afriendcalledben, 2017-01-16:
So sorry, ignore me. 
Was not using python3 to run the scripts and failed to realise they were already running. 
Again, love this work!

closed 2017-01-16 by afriendcalledben

adangert, 2017-01-17:
Thanks so much!
running setup.sh should get everything working right out of the box!
anyway, keep on the lookout for more games/modes coming soon!! There is lots in store for JoustMania :)