adangert/JoustMania #35: mp3 support

adangert, 2017-02-03:
right now, wavs are supported, mp3 support needs to be added and tested, This might be difficult with the real-time resampling in piaudio.py, perhaps quickly convert the mp3 to a temp wav first before the game starts then delete it after the round is over?

ericbuzan, 2017-08-25:
I poked around a little bit and sticking with wav seems to be the best bet without a ton of coding. I'd whip up a script ([pydub looks good](https://github.com/jiaaro/pydub/)) that quickly converts typical formats into wavs and just save those. Maybe have that script run on the included song to save some download time?

adangert, 2017-08-25:
Yeah that probably would be the best solution for now.

adangert, 2017-08-29:
now supports mp3, flac, ogg and others http://www.ffmpeg.org/general.html#File-Formats

closed 2017-08-29 by adangert