adangert/JoustMania #43: players turn red when they die

adangert, 2017-05-08:
To make the game have better player feedback, when a player dies, their controller should flash and turn red. The red in the other game types for player colors will have to be changed as well (FFA, Teams, Werewolfs, Zombies, commander) so that players are not confused

MrAbodi, 2017-05-31:
is red really superior to off? i don't see how. also how if at all would that affect the battery life.

adangert, 2017-05-31:
red is very ingrained as stop in society, plus when people die I've seen a couple that continue for a second or two because they don't notice their controller turned off. The red might be more visible. Also I haven't really seen the battery die with any of the controllers I've been playing with, so I'm guessing it would be negligible.

MrAbodi, 2017-07-04:
Hi Aaron, Another consideration with regards to this.

This past weekend we played the game in very low lighting, the light turning off in this instance is much more evident of dying then a change of color would be. 

Had a brilliant time BTW. I bought another bluetooth adapter and 2 more moves. ;)

adangert, 2017-10-17:
This has been updated in the new version, currently there is an option to make the controller turn red or dark when they die, although I am slightly leaning towards red.

closed 2017-10-17 by adangert