adangert/JoustMania #50: what about a king of the hill mode.

MrAbodi, 2017-05-31:
I was thinking about a **King of the Hill** mode, which is probably better suited to a smaller number of players, but maybe not.

Something along of the lines of everyone starts neutral. once someone is knocked out, randomly from the remaining players someone becomes the King, and everyone remaining else is one color (red?). players who die respawn after a certain amount of time. 

While you are King, you accrue points over time, the current King wins once he has accrued a certain number of points. (point total could be varied depending on desired game length)

If the King is joustled, he dies awaiting respawn and another random player becomes King. 
Perhaps it's better to say only alive player could become king, resulting in some fun backstabbing between non-kings players to improve their odds of becoming King.

Requiring playtesting. i'm not sure whether to win you must accrue those points in a single shot of being King (might be too hard or limiting) or over the course of a match. 

If people are interested in how they went, at the end the controllers could go a unique color and the game could say the score.  
"Green 20... Red 18... Yellow 5... etc"

thoughts?

adangert, 2017-05-31:
I like the idea, there would need to be balancing with the sensitivity though, i.e. if it's 13 vs 1 it's going to be very quick before the king dies, perhaps everyone else has a lowered sensitivity, but yeah it would require some playtesting. I like the idea though! 👍

adangert, 2017-05-31:
If you want to make a branch to create and test it out, go for it!

MrAbodi, 2017-06-01:
i'm thinking about it :) but i've never programmed anything like this before.  Mostly Pascal back in the 90's and VBA macros in excel/access these days.

MrAbodi, 2017-06-01:
Decided to give it a go, but how do i create a branch? sorry i'm a total noob with github.

adangert, 2017-06-01:
haha no problem, you probably want to fork first though then branch if you want. here is a good guide on it: https://guides.github.com/activities/forking/