adangert/JoustMania #53: LED lag since recent update

MrAbodi, 2017-06-04:
I'm not sure it somehow it's just my setup, but since updating recently. I've encountered 2 issues that i was not experiencing previously. 

1) I'm seeing controller LED's all turn off and need to reset to get the controllers to respond agaon.
2) during play. joustling somebody is working, and i can hear the sound effect to tell me they are out, it's prompt and so i don't believe it to be a bluetooth issue. 
however when someone dies, the LED goes white for maybe 5 seconds then goes out or the controller just continues to be on for 5 off seconds before going out.

anyone else experiencing this? or not?

adangert, 2017-06-04:
I'll check to see if that is happening, usually for 1 it means the pi was overheating causing the pi to restart, it the flask web interface might be putting load on the pi. Number 2 sounds like an issue with the new sensitivity detection.

MrAbodi, 2017-06-04:
hmm heat issues, it's not unlikely. but i had the lid of the case off. meaning active cooling will be a necessity

adangert, 2017-06-05:
yeah I've seen when the crash sporadically, when it is using all it's resources, i.e. all 16 controllers going, I've also seen the pi crash when it's just overheated and joust is going too. So that might be an issue. I have some small heat syncs on the pi which seem to help a bit. Also I've updated Joust to take care of the LED lag you were seeing, if you do a git pull those issues should go away.

closed 2017-06-05 by adangert

adangert, 2017-06-05:
Here is the previous issue to the pi shutting off: https://github.com/adangert/JoustMania/issues/24

MrAbodi, 2017-06-05:
Great. I'll give it try tonight.