adangert/JoustMania #55: Waves mode

adangert, 2017-06-05:
One player is the Juggernaut, there are different waves where players from the sidelines spawn to attack the main player. Main player has bombs he can use, to kill all enemies. The main player also progresses through different waves, and can reach checkpoints which are saved by a pin code from the controller. 

--extra detail: enemies: red to green for difficulty, purple for invincibility.

adangert, 2017-09-16:
Perhaps for simplifying, 1v1 in a group, 10 seconds otherwise both die. First to 5 points wins.

Robotopera, 2017-11-02:
Could be 1v1 with a group in a circle fight club style where the winner advances and the loser gets put at the back of the line. 
Game starts 2 controllers illuminate, player on deck vibrates. After a win the losing players gets sent to the back of the lineup, the player on deck's controller lights up to fight and a new players controller vibrates to indicate they are on deck. Who ever has the highest number of wins after x rounds wins.

adangert, 2017-11-05:
Yeah I think that would work well. Although I think it would still be beneficial to keep the 10-15 second rule in there (a bunch of unbounded 1v1 fights would take forever), perhaps with a warning before both players die. 

Some neat ideas to consider, perhaps if the player on deck is on a long win streak (i.e. 5+) then start throwing two players at them. Also when the game finishes then announce the winners win streak, perhaps if the winner doesn't have the highest win streak, then announce both highest score and win streak?

adangert, 2018-01-04:
Fight club mode added will need to be playtested though.

closed 2018-01-04 by adangert