adangert/JoustMania #67: get Joustmania working on Windows

adangert, 2017-06-24:


adangert, 2017-08-27:
https://github.com/thp/psmoveapi/pull/313 was just fixed, this may make it easier to implement joustmania on windows.

adangert, 2017-09-04:
Currently is blocked by some issues with psmove on Windows, here: https://github.com/thp/psmoveapi/issues/314 and here: https://github.com/thp/psmoveapi/issues/214

mbabinski-at-google, 2017-09-20:
FWIW, JS Joust was never released on Windows because Microsoft kernel engineering claimed that the bluetooth stack doesn't support what's required.

https://arstechnica.com/gaming/2014/12/microsoft-tells-j-s-joust-devs-their-game-is-not-possible-on-windows/

martinbogo, 2017-11-07:
@adangert The PSmove just won't work on the Windows bluetooth stack ... even the newest win10 creator enhancements don't help.

adangert, 2018-06-01:
There is a dll that seems like it can be implemented: https://github.com/thp/psmoveapi/issues/355

adangert, 2019-05-22:
updated and working in windows branch.

closed 2019-05-22 by adangert