adangert/JoustMania #68: Joustmania on Pocket Chip

adangert, 2017-06-24:
the pocket Chip has built in bluetooth (perhaps not class 1) and a screen, it would be really nice to have this as a super portable JoustMania machine.

Leyart, 2017-07-21:
I would really love to help you with this; i don't think that the chip internal bluetooth is strong enough for this but we should try. Do you own one? I do.

adangert, 2017-07-21:
I don't own one yet, although I probably should get one, just in case this works out (I suspect it will). It would be nice to try to it working with the internal bluetooth, even though it probably won't really work with more than 4 controllers at close range, however I suspect as well that inserting a class 1 bluetooth dongle into the usb port will help with it's shortcomings. If there would be a way to get the audio to work through the built in port that would be fantastic as well, although a usb hub probably wouldn't be too much of a compromise (or the addition of the Joust silent mode might work as well) 

If you want to start tinkering with a solution go for it. I'll see about getting a pocket chip soon.

adangert, 2017-07-21:
Couple things to note as well, Psmoveapi support was recently added for the Pocket Chip [Link](https://github.com/thp/psmoveapi/blob/master/docs/build.rst#building-for-the-pocket-chip) so getting it to work shouldn't be that much of a hassle, also bonus points if the some gui, or the JoustMania webapp, could be displayed on the chip screen and be controlled from its keyboard

Leyart, 2017-07-21:
The only reason to make Joustmania work on the Pocketchip would be to have an admin interface directly on the screen, that would be the major goal. I am unfortunately extremely busy with work, so I won't be able to start working on it earlier than mid October but I will reach you out as soon as I will put my hands on it.

adangert, 2017-07-21:
I would say that and the ability for a more portable Joustmania setup, i.e. it has the battery built-in.

thp, 2017-09-03:
Should be easy to do, the Pocket C.H.I.P is mostly a Debian ARM box, does have 3.5mm audio out, a full-size USB host port and built-in WiFi and Bluetooth. The Bluetooth isn't class 1, but one could use a USB dongle for faster BT. For GUI, WebUI is probably a bit too much for the C.H.I.P, but X11 + EGL + OpenGL ES 2.0 can give a nice UI.

martinbogo, 2017-11-07:
Sadly, the Class 1 BT USB device I got from Amazon pulled too much current for the little C.H.I.P.   and causes instability.   Also having some serious issues with pyaudio... 

"Damnit Jim, we need more CPU cycles!"