adangert/JoustMania #73: Silent Mode?

hdsjulian, 2017-07-18:
I'd love to see a "silent mode" that allows me to play the game in a park or at a festival where music isn't an option. This would require to not only "not play music" but also not having any kind of speed-variance throughout the game because there would be no indication of these speed changes through the music.

adangert, 2017-07-20:
That's not a bad idea, and actually shouldn't be too hard to implement

adangert, 2017-07-22:
Updated joustmania with the change, change `audio = false` in joustconfig.ini to turn on this mode. Werewolves and Commander are not playable as they need audio to work with the game. Also the sensitivities for the controllers is set to (FAST_MUSIC_SPEED + SLOW_MUSIC_SPEED)/2 for each game that has the speed changes.

closed 2017-07-22 by adangert