adangert/JoustMania #88: Research using phones as controllers

adangert, 2017-08-29:
It is probably possible to use your phone as a controller through the webui if you run out of move controllers to pass around. This way more people could join in if you don't have enough move controllers.

adangert, 2017-08-29:
This might also open the door for other game types as well. In the same vain as the Jackbox party TV games, however no screen required.

adangert, 2018-05-10:
WebBluetooth is now released, perhaps look into getting this to work with phones: https://webbluetoothcg.github.io/web-bluetooth/

adangert, 2019-05-06:
With the AP, I don't think that there should be a limit to the amount of phones that could connect, therefore it's probably a better solution vs the 7 controllers per bluetooth dongle, it will remain to be seen how many phone a single pi can support over wifi however.

adangert, 2019-05-06:
Look into https://github.com/noffle/proust

adangert, 2019-05-06:
also http://vitor.io/designing-johann-sebastian-joust-html5-javascript

supersmo, 2019-06-06:
Found this as well: https://github.com/bladechapman/Joust