adangert/JoustMania #90: Kernel oops while syncing controllers

ericbuzan, 2017-08-29:
This is from a fresh install of Raspbian Stretch and Joustmania.
```
Message from syslogd@raspberrypi at Aug 29 20:05:20 ...
kernel:[  133.361578] Internal error: Oops: 17 [#1] SMP ARM
kernel:[  133.458312] Process python3 (pid: 666, stack limit = 0xaa56c210)
kernel:[  133.464313] Stack: (0xaa56ded0 to 0xaa56e000)
kernel:[  133.468663] dec0: aa56df4c 80c09474 0000080f 8071abf8
kernel:[  133.476834] dee0: 7ee04510 aa565e78 9e95f480 c0104807 aa56df7c aa56df00 80283730 805e6b5c
kernel:[  133.485005] df00: 00000000 00000001 b13f3e48 00000000 a9865c20 802860dc 00000000 00000000
kernel:[  133.493178] df20: 00000000 00000015 b13f3e48 00000000 a9865c20 8028fcc4 b1010500 0262ab78
kernel:[  133.501349] df40: 7ee04510 7ee04530 c0104807 00000016 aa56df6c 9e95f481 7ee04510 9e95f480
kernel:[  133.509520] df60: c0104807 00000016 aa56c000 00000000 aa56dfa4 aa56df80 80283ee8 80283690
kernel:[  133.517692] df80: c0104807 0262ab78 00000000 7ee04530 00000036 80108244 00000000 aa56dfa8
kernel:[  133.525863] dfa0: 801080c0 80283eb0 0262ab78 00000000 00000016 c0104807 7ee04510 c0104807
kernel:[  133.534034] dfc0: 0262ab78 00000000 7ee04530 00000036 769a7570 769599d0 00378000 6b820df0
kernel:[  133.542205] dfe0: 7698e358 7ee04504 76965da8 76da38ac 60000010 00000016 00000000 00000000
kernel:[  133.572919] Code: e7f33053 e79c3103 e5935018 e5953e0c (e5933020)
```

adangert, 2017-08-30:
also referenced here: https://github.com/adangert/JoustMania/issues/34

adangert, 2017-09-02:
Might want to try something like this: https://unix.stackexchange.com/questions/290336/safely-remove-usb-from-linux-device