adangert/JoustMania #91: Safe Shutdown function

Nam-Ereh-Won, 2017-09-05:
Since this is intended to be headless, you don't have many options for starting a safe shutdown outside of rigging up a button and setting up a trigger. Perhaps holding the Move and Trigger buttons on a controller in admin mode could initiate a safe shutdown and/or a button in the Web Interface.

adangert, 2017-09-05:
Might be nice to have in the web interface, although I've had no problems at all with just pulling the power on the pi though, the whole system seems to be very resilient.

mbabinski-at-google, 2017-09-15:
It might be more or less sufficient to just flip on the "sync" mount option on "/". It will make writing to disk slower, but JoustMania shouldn't be doing much writing.

ericbuzan, 2017-10-07:
My latest pull request has options to reboot or shutdown in web settings. I would like to add a button combination as well, but we gotta figure out the right combo that "drunk player randomly mashing buttons" has a minuscule chance of tripping.

adangert, 2017-10-09:
Holding both start and select for a couple seconds probably would do the trick.

martinbogo, 2017-11-07:
How about a two-stage combo for shutdown and/or reboot:

- hold triangle + select for 5 seconds ( shutdown )
- hold circle + select for 5 seconds ( reboot )
- vibrate controller(s) and set to blinking red, announce "shutdown" or "reboot" via sound or verbally
- trigger confirms shutdown or it times out within another 5 seconds.