adangert/JoustMania #123: Allow any game to start when all players press trigger, not just random mode.

adangert, 2017-10-09:


Robotopera, 2017-11-02:
Also allow players to force start a game by x number players holding trigger to start count down. Useful if someone wants to take a break.

adangert, 2017-11-03:
you can push all four front buttons to turn off a controller currently.

martinbogo, 2017-11-07:
Aaron,

I agree, that the game mechanic would be easier/more intuitive with "trigger-in" vs "opt-button-out" as follows:

*) Admin selects game, and presses <start>
*) Game type is announced ( "Free For All" )
*) If enabled, Instructions are played
*) Admin configurable time-out ( 10 seconds ) with announcement "Press trigger to join in"
*) Color cycle intro starts, game begins.

adangert, 2017-11-08:
yeah opt-in is good, especially for multiple players, but the Admin shouldn't always have to start a new game, i.e. the system should be able to run itself with just the players currently in. Perhaps after a game ends, once the first player triggers in, then like a silent minute countdown starts,(also giving players to take a breath from the previous game). at the 10 second mark the press trigger to join announcement is activated and all the controllers that have not been triggered are out of the next game.

adangert, 2017-12-29:
update the game for the main issue, i.e. all games require a trigger in now., default game is now FFA rather than random mode.

closed 2017-12-29 by adangert