adangert/JoustMania #124: Re-evaluate how controller motion is measured.

mbabinski-at-google, 2017-10-16:
The current implementation sums the components of the acceleration vector returned by the move hardware. This has several undesirable properties:

1. It is biased: the controller reports a downward accelaration of 1G even when it is perfectly still.
2. It is not orientation-invariant: an acceleration along a sensor axis will produce a reading roughly 2/3 as strong as an acceleration in the x-y plane 45degrees off of either axis
3. As a special case of #2, rotating the controller affects the sign of the components. In an extreme case, a lateral acceleration of 1G could either cancel with gravity (leading to zero motion registered), or *double* it, leading to a quick knock-out.

Fixing #2/#3 is easy in principle: compute a magnitude (or at the least, a sum of absolute values). 

De-biasing is harder: conceivably the magnetometer could be used to compute "down" in order te de-bias acceleration, but it requires calibration, and the values might themselves be biased due to local conditions. My experimental FFA game calculates [Jerk](https://en.wikipedia.org/wiki/Jerk_(physics)), and that looks promising, but requires play-testing and finding reasonable values.

mbabinski-at-google, 2017-10-16:
@adangert how did you come up with acceleration thresholds in the game right now? Trial and error and playtesting?

adangert, 2017-10-17:
yeah just playtesting until it felt right, both indoors and outdoors with multiple players, also made sure that the vibration of the warnings felt right, and didn't trigger players out if they were holding the controller too softly.

adangert, 2017-10-17:
the above suggestions sound great btw.

martinbogo, 2017-11-07:
@mbabinski-at-google Do you have a reference code example for the jerk calculation?   I may have stumbled onto a way to do a magnitude along a vector, thanks to the code to do acceleration from delta 3D printers.

adangert, 2019-05-29:
The algorithm is now the sum of the squares for the x, y, and z acceleration vectors. Also now it uses a much simpler and better moving average algorithm.

closed 2019-05-29 by adangert