adangert/JoustMania #132: Cannot install JoustMania on latest Rasbian Stretch Lite

martinbogo, 2017-11-01:
I've been trying to install JoustMania on an Rpi3 using Raspbian Stretch Lite ( Sept 2017 edition ) and having no luck.    Joust does not start up... it throws a numpy error when I attempt to manually start Joust. 

pi@raspberrypi:~/JoustMania $ sudo ./joust.sh 
Traceback (most recent call last):
  File "/home/pi/JoustMania/piparty.py", line 2, in <module>
    import common, colors, joust, webui
  File "/home/pi/JoustMania/joust.py", line 8, in <module>
    from piaudio import Audio
  File "/home/pi/JoustMania/piaudio.py", line 8, in <module>
    import scipy.signal as signal
  File "/usr/lib/python3/dist-packages/scipy/__init__.py", line 114, in <module>
    from scipy._lib._ccallback import LowLevelCallable
  File "/usr/lib/python3/dist-packages/scipy/_lib/_ccallback.py", line 1, in <module>
    from . import _ccallback_c
ImportError: /usr/lib/python3/dist-packages/scipy/_lib/_ccallback_c.cpython-36m-arm-linux-gnueabihf.so: undefined symbol: PyFPE_jbuf

Here are the current installed Python3 libraries:

ii  libpython3-dev:armhf               3.5.3-1                        armhf        header files and a static library for Python (default)
ii  libpython3-stdlib:armhf            3.5.3-1                        armhf        interactive high-level object-oriented language (default python3 version)
ii  libpython3.5:armhf                 3.5.3-1                        armhf        Shared Python runtime library (version 3.5)
ii  libpython3.5-dev:armhf             3.5.3-1                        armhf        Header files and a static library for Python (v3.5)
ii  libpython3.5-minimal:armhf         3.5.3-1                        armhf        Minimal subset of the Python language (version 3.5)
ii  libpython3.5-stdlib:armhf          3.5.3-1                        armhf        Interactive high-level object-oriented language (standard library, version 3.5)
ii  libpython3.6:armhf                 3.6.3-1                        armhf        Shared Python runtime library (version 3.6)
ii  libpython3.6-dev:armhf             3.6.3-1                        armhf        Header files and a static library for Python (v3.6)
ii  libpython3.6-minimal:armhf         3.6.3-1                        armhf        Minimal subset of the Python language (version 3.6)
ii  libpython3.6-stdlib:armhf          3.6.3-1                        armhf        Interactive high-level object-oriented language (standard library, version 3.6)
ii  python3                            3.5.3-1                        armhf        interactive high-level object-oriented language (default python3 version)
ii  python3-apt                        1.1.0~beta5                    armhf        Python 3 interface to libapt-pkg
ii  python3-cffi-backend               1.9.1-2                        armhf        Foreign Function Interface for Python 3 calling C code - runtime
ii  python3-crypto                     2.6.1-7                        armhf        cryptographic algorithms and protocols for Python 3
ii  python3-cryptography               1.7.1-3+b2                     armhf        Python library exposing cryptographic recipes and primitives (Python 3)
ii  python3-dbus                       1.2.4-1                        armhf        simple interprocess messaging system (Python 3 interface)
ii  python3-decorator                  4.0.11-1                       all          simplify usage of Python decorators by programmers
ii  python3-dev                        3.5.3-1                        armhf        header files and a static library for Python (default)
ii  python3-gi                         3.22.0-2                       armhf        Python 3 bindings for gobject-introspection libraries
ii  python3-idna                       2.2-1                          all          Python IDNA2008 (RFC 5891) handling (Python 3)
ii  python3-keyring                    10.1-1                         all          store and access your passwords safely - Python 3 version of the package
ii  python3-keyrings.alt               1.3-1                          all          alternate backend implementations for python3-keyring
ii  python3-minimal                    3.5.3-1                        armhf        minimal subset of the Python language (default python3 version)
ii  python3-numpy                      1:1.13.1-1                     armhf        Fast array facility to the Python 3 language
ii  python3-pip                        9.0.1-2+rpt1                   all          Python package installer
ii  python3-pkg-resources              33.1.1-1                       all          Package Discovery and Resource Access using pkg_resources
ii  python3-pyasn1                     0.1.9-2                        all          ASN.1 library for Python (Python 3 module)
ii  python3-scipy                      0.19.1-1                       armhf        scientific tools for Python 3
ii  python3-secretstorage              2.3.1-2                        all          Python module for storing secrets - Python 3.x version
ii  python3-setuptools                 33.1.1-1                       all          Python3 Distutils Enhancements
ii  python3-six                        1.10.0-3                       all          Python 2 and 3 compatibility library (Python 3 interface)
ii  python3-wheel                      0.29.0-2                       all          built-package format for Python
ii  python3-xdg                        0.25-4                         all          Python 3 library to access freedesktop.org standards
ii  python3.5                          3.5.3-1                        armhf        Interactive high-level object-oriented language (version 3.5)
ii  python3.5-dev                      3.5.3-1                        armhf        Header files and a static library for Python (v3.5)
ii  python3.5-minimal                  3.5.3-1                        armhf        Minimal subset of the Python language (version 3.5)
ii  python3.6                          3.6.3-1                        armhf        Interactive high-level object-oriented language (version 3.6)
ii  python3.6-dev                      3.6.3-1                        armhf        Header files and a static library for Python (v3.6)
ii  python3.6-minimal                  3.6.3-1                        armhf        Minimal subset of the Python language (version 3.6)

Robotopera, 2017-11-02:
Did the setup script complete without any errors?

martinbogo, 2017-11-02:
Yep.. completed w/o errors.  I just tried to install on Raspbian Stretch Desktop, with the same issue.

martinbogo, 2017-11-02:
I'm going to try building scipy rather than using the one installed by debian with pip:

    PYTHON=$VENV/bin/python3.6
    $PYTHON -m pip install --ignore-installed psutil flask Flask-WTF pyalsaaudio pydub pygame pyaudio pyyaml dbus-python || exit -1
    $PYTHON -m pip install --upgrade --force-reinstall scipy

Not sure this will help... but something is either conflicting, or wrong, in the venv.

Robotopera, 2017-11-02:
[According to this it could be caused by multiple versions of numby being present on your device.](https://stackoverflow.com/questions/36190757/numpy-undefined-symbol-pyfpe-jbuf)

Are you starting from a clean install?

adangert, 2017-11-02:
I will try a clean install and see if anything happens.

adangert, 2017-11-02:
there was some weird error with multiple numpy installations. The new setup.sh should work, although it will be slower. I'll try other solutions tonight.

closed 2017-11-02 by adangert

martinbogo, 2017-11-03:
awesome!  I did have to rebuild scipy to make things work.   took about 45
minutes.

adangert, 2017-11-03:
yeah it seems the problem is with numpy rather than scipy, I'm trying to fix the install script