adangert/JoustMania #141: Add support for 3D printed / Open Source ESP8266 "m00ve" controller

martinbogo, 2017-11-07:
I'm working on an open source, 3D printed controller called the "m00ve" that has some of the same features as the PS3 move controller.  

It is WiFi ( longer range, low latency ), uses a standard 18650 liION battery for power, one NeoPixel for color on top and a standard ADXL 9DOF chip for accelerometer etc.

Only six buttons though ( trigger + five arranged in a domino - 5 pattern ).  No easy way for me to expand to more because of the lack of pins on the ESP chip.

adangert, 2017-11-08:
yeah that sounds fantastic! You might want to look here for some inspiration as well: http://www.opentilt.org/

adangert, 2019-05-30:
Just saw these on twitter: https://twitter.com/puseyscience/status/1131894862343577601