adangert/JoustMania #155: fourth sensitivity setting?

adangert, 2018-01-08:
Between medium and low (perhaps this could be the default, and the other send sensitivities removed?)

ChainmailBuddha, 2019-04-03:
After recently playing this with a large group of experienced players (all with extensive experience with the original JSJ) the single most common feedback was that the game was not sensitive enough. At a minimum we need to have one or more settings that are significantly more sensitive than the current options.

I think it might also be good to have the option of a bit more granularity, at least in the web interface.
e.g. have 3-4 settings available in headless mode (with the most sensitive being much more sensitive that now) and perhaps a slider in the web interface with 5-10 options which not only extend past the options in headless mode, but offer better fine tuning.

adangert, 2019-05-06:
Added five different sensitivity levels, with much better variability between them. Also changed the sensitivity detection algorithm for a better play experience.

closed 2019-05-06 by adangert