adangert/JoustMania #156: joust broke WIFI After Running enable_ap.sh or disable_ap.sh

Ryan-unity1993, 2018-01-10:
after running one of the ap.sh's i lost my wifi. (No wireless Interfaces).
here is how to get your WIFI Back. go to "/etc/dhcpcd.conf" and open it, look for "#Access point for JoustMania" Change the line below that to "interfaces wlan0" your wifi will come back.

adangert, 2018-01-11:
that's right, wifi will stop working when you run enable_ap.sh, it should start again after disable_ap.sh though.

adangert, 2018-01-11:
added this line to the README, we should look into this and build it into the disable_ap.sh if needed.

Ryan-unity1993, 2018-01-14:
after running the disable_ap.sh my WIFI did not come back. I went in and fixed it. and posted how to fix.
you will need to run the disable_ap.sh after reboot you will need to implement the fix posted above to fully return WIFI.

adangert, 2019-06-06:
this shouldn't be broken anymore.

closed 2019-06-06 by adangert