adangert/JoustMania #157: Bluetooth Sound Support?

Ryan-unity1993, 2018-01-14:
I would like to have Bluetooth sound to have less wires coming from the RPi3 and can play the sound from more than 2 speakers.

adangert, 2018-01-17:
it's a good idea in theory, although that may cause more congestion on the bluetooth channels, and thus less accurate gameplay with the controllers. This should be optional to implement.

Ryan-unity1993, 2018-01-18:
I was thinking designated Bluetooth out put. like if you have more then one Bluetooth USB stick. the on bored Bluetooth can run audio only. or set one of the USB Bluetooth to run audio on maybe up to 4 Bluetooth speakers. my pi has a 3.5 LCD. less is more. I would like to be able to run something like surround sound without all the wires for the play zone. it would be easier to setup and I would not have to worry people will trip on wires when they get a little trippy and tipsy.

adangert, 2018-01-18:
yeah as long as it doesn't cause extra problems with the gameplay, i.e. more processing power. I also run a 16 controller version of the game, and extra bluetooth interference could possibly be detrimental to the controllers running correctly. If you want to build in a script to have it run on bluetooth you're more than welcome. :)

Wowfunhappy, 2018-05-18:
Speaking from experience with the official version of Joust: attempting to use Bluetooth audio alongside a substantial number of Move controllers always ends poorly. This happens even if I'm using external, class 1 Bluetooth adapters _and_ even when I use a separate 3.55mm-to-Bluetooth adapter like the one linked above. I assume there's just too much congestion in the airwaves, regardless of the hardware in use.

adangert, 2018-06-02:
going to close this for now, since bluetooth is the main way of communicating with the speakers, extra congestion probably is not a great idea.

closed 2018-06-02 by adangert