adangert/JoustMania #158: Problems with re-connecting Move controllers after using with Playstation

woodcoder, 2018-02-02:
We've seen an issue where after having successfully used a set of Move controllers to play JoustMania, but then connected them to a Playstation (e.g. to charge) they then no longer work with the Pi and it seems to be impossible to get them to pair again without resetting everything.
____

This has been observed with the version in the repo on 7 Oct 2017, and also with a `git pull` / `sudo ./setup.sh` upgrade today.  (i.e. this is not a completely clean install.)  We're using three external Trendnet bluetooth adaptors.
____

The only reliable procedure to get all the controllers paired and working properly is to:

   * plug the Pi into a screen and start a terminal
   * run:

```
sudo rm -fr /var/lib/bluetooth/*
cd JoustMania
./reset_bluetooth_connections.sh
```

   * this will reboot the machine

While it is rebooting:

   * make sure all the external bluetooth adaptors are connected (you’ll need to swap out keyboard/mouse)
   * wait for all of the blue lights on the adaptors are flashing (machine is running properly)
   * in turn, plug each move controller into the USB port and wait for it to glow white and then switch off (_do not_ press the playstation button yet*!)
   * once all controllers have been through this process, then wake them all up using the playstation button, they should all turn blue

If some of them don’t connect, reboot (power off/on) and repeat the connection stage (the second set of steps).

Then it’s good to go!

[* I think the registering phase gets interrupted if you press the playstation button, because that starts the game code?]
____

The problem seems to come if you connect any of the controllers to a Playstation.  Afterwards, that controller will no longer recognise the Pi — even if you plug into the Pi’s USB port again (it will turn white, but then when you disconnect and press the playstation button it will fail to connect with a rapid flashing red light).

The only way I’ve managed to get it to reconnect is by resetting the whole system as above, and reconnecting every controller.

Having looked at the contents of /var/lib/bluetooth, it appears that this is a directory that contains folders named after the bluetooth address of each adapter (they look like 00:00:ba:dc:12:34 or similar).  Inside each of these is a subfolder named after the bluetooth controller of the move controller that’s registered with it (again, something like 00:00:ef:56:78:90).

I think what’s happening is that when you plug in the move controller for the second time (after registering it successfully, and then plugging it into a Playstation) it is recognised by the Pi and failing to somehow re-pair the controller with the Pi (because it doesn’t know that it needs to do that because it doesn’t know about the Playstation).  Running the rm clears out these folders and so it all works if you start again from scratch.  Just running the most recent `./reset_bluetooth_connections.sh` script does not seem to be enough.  I presume that if you knew the bluetooth address of the affected controller, you could probably just remove its subdirectory to get this to work too.

The final thing I’ve noticed is that it’s important to have all the bluetooth adaptors connected when you start registering the controllers as they seem to be randomly (possibly round robin) allocated to the adaptors — therefore if you don’t have them plugged in, then all the controllers get registered to the one adapter.  I don’t know if it’s possible to make the controllers register with all of the adaptors, although that would seem desirable if the protocol allowed that / it led to a stable system.
____

Would it be solved by a completely clean install -- or is there a better way to solve this problem?

adangert, 2018-02-04:
good observations, you are correct about most things, it pair round robin to each of the connected adapters, and yes you need to run reset_bluetooth_connections.sh if one controller becomes out of sync, and sync all the controllers again.

A good solution that would need coding, is once a controller is plugged in, check to see if a folder does not exists, if it does, then somehow it would need to check if that folder is still valid (or perhaps check if the settings on the playstation remote are still valid). If they are not, delete the folder and re-sync.

I don't know how difficult the above would be, but if the psmoveapi does not allow looking into the controller itself for the bluetooth settings, it might be very hard or technical to code up. A very simple work around is just to charge the controllers with the raspberry pi! This way nothing becomes out of sync, and it's relatively easy for storage. There is a hub that works with charging the controllers on the main readme.

closed 2018-02-04 by adangert

woodcoder, 2018-02-05:
Thanks for the confirmation of the behaviour.  I did find that I needed the `rm` in addition to the reset script, but it then does work reliably.  We'll stick with the workaround of not connecting the controllers to a Playstation.

Many thanks to you (and your contributors) for a great set of games!

adangert, 2018-02-05:
oh yeah, you might need to do sudo -i first before the script in order for it to rm the directories.