adangert/JoustMania #169: Feature Request - Normal Speed during Gameplay

Nam-Ereh-Won, 2018-07-09:
Currently the game only plays back music at high or low speed, but not at normal speed. The original JS Joust functioned with the music playing at normal speed and either speeding up or slowing down.

Shouldn't be too difficult to add that feature in.

adangert, 2018-07-09:
So it starts at normal speed, and either goes high or low and back again? I'll have to check out videos of the original to verify.

ChainmailBuddha, 2018-07-10:
If we are making feature requests for music tempo, I would also love to see the same freeze functionality from the original added into this port as well 😄

Toastytwo, 2018-07-10:
Hi, normal speed is in J.S. Joust regularly, noticed that difference to Joustmania too!
You're just amazing, love you for making it portable!

Nam-Ereh-Won, 2018-07-11:
I got normal, fast, and slow speeds working in my fork. The sensitivity and duration values need tweaking, but otherwise it works just fine. It'll start normal speed and randomly switch to high or low for a while, then back to normal speed, ect ect. I don't think Freeze would be any more difficult. Looks like someone else was working on freeze and normal speed judging by some things I found poking around.

I've got another feature I'm working on that might've broken support for music without metadata, but once I fix that I can see about doing a pull request. Also squashed a bug where the menu music stops when attempting to start a game without the minimum amount of players.

Toastytwo, 2018-07-11:
Wow, sounds great, will try your fork for sure! Could you give me a hint where to look for the timing/sensitivity settings? I have a small event at work in a few days and would like to get the additional speed setting(s) right!

adangert, 2018-07-11:
Yeah these are all great suggestions! The only thing is that freeze mode should be able to be turned on or off, I know some people that do not like it as well.

Nam-Ereh-Won, 2018-07-11:
I figured out why files without metadata broke and fixed it. As for the sensitivity, common.py has the settings in it and joust.py has the the segment duration. There are comments to help you out. I need to do some reading up on AJAX as I'm having some difficulties with the tweaks I want to do that need stuff added to the WebUI.

ChainmailBuddha, 2018-07-11:
@adangert I completely understand and agree with you on freeze. I would even be fine if it is a setting buried in the config/webui like the no music mode.

It is not a critical thing, but something some of the groups I play with are used to and enjoy from the original.

Nam-Ereh-Won, 2018-07-11:
I'll get to adding freeze once I refactor the game modes to use inheritance so it's easier to update the common stuff.

Toastytwo, 2018-07-14:
I know, it's OT, but can someone help me out fixing a quirk? Got YM working, jesse-lite, USB soundcard and like 6 moves. Worked fine but sometimes instead of starting a new round (FFA or Teams mainly) the move controllers just turn off the glow ball and are not responding to trigger or team select buttons anymore. They keep paired but a reboot is necessary to get YM running again. The wireless AP is activated, but the error happens also when using the controller inputs to config and start a game.

adangert, 2018-07-14:
Looks like it might be related to this issue: https://github.com/adangert/JoustMania/issues/163

It could also be the power supply you are using to power the pi.

ChainmailBuddha, 2018-07-14:
@Toastytwo @adangert I have had some similar glitchy issues such as this. Some games fail to start even though everyone has hit their trigger, and the webui shows all players active. Usually I can force start the game from the webui, but sometimes the reboot is required.

I also considered power to be a possible issue, except I have had these issues both with a power adapter from the wall, as well as running from a solid and reliable power bank.

As for the os version, I am running the latest fully updated Stretch, so I don't think that Jessie-lite is the issue.

We should likely open another issue for this @Toastytwo but I wanted to let you and @adangert know this was not an isolated issue.

egburr, 2024-10-02:
> Hi, can confirm normal speed is in J.S. Joust regularly, noticed that difference to Joustmania too! And there was an announcement saying "FREEEZE!!!" and the music stopped for a few seconds, was great fun! You're just amazing, love you for making it portable!

Also with freeze mode, the controllers would all turn light blue for 1-2 seconds at the "freeze!" announcement, as a visual indicator.