adangert/JoustMania #172: Pairing not using additional Bluetooth Adapters

Nam-Ereh-Won, 2018-07-15:
I purchased a second BT adapter to support more players, but the game doesn't want to use it. If I have either one plugged in it'll use it, but it will only use one if both are connected. The jm_dbus and pair files seem to indicate that it SHOULD pair a controller to whichever adapter has fewer moves on it, but this does not seem to be the case. I've ran the reset bluetooth script, but still getting the same behavior as I repair the controllers. I'm tempted just just reinstall from scratch, but that shouldn't be necessary.

It may be important to note that I'm using two of the same brand and model adapter.

adangert, 2018-07-15:
I bet both adapters have the same Mac address, you can check if you go to /var/lib/Bluetooth

Nam-Ereh-Won, 2018-07-15:
Yup, making the dev version of bluez so I have bdaddr and we'll see what happens next.

Nam-Ereh-Won, 2018-07-16:
Could be handy to include some troubleshooting tips in the wiki.

adangert, 2018-07-16:
Added

closed 2018-07-16 by adangert

nigelkennington, 2023-05-07:
How is it possible to know if the controllers are only pairing to one Bluetooth adapter? I'm only able to connect 5 and then my remaining 3 wont connect even after apparently pairing correctly (connecting with a wire and the light going white).

Also, it might be worth mentioning that the instruction "go to /var/lib/Bluetooth" is very unclear for linux newbies. To actually see the mac addresses, you need to do something like this in the terminal:

1. sudo bash
2. cd /var/lib/bluetooth
3. ls

That should give you a list of numbers like this:

00:15:83:F4:A1:B5  00:15:83:FB:C6:7F  00:1A:5D:DA:71:13

That's 3 different mac addresses separated by spaces.

DrDecagon, 2023-07-19:
For users looking for another option to solve this problem.  bdaddr is not included in debian builds of bluez but bccmd is included.  bccmd only works on Cambridge Silicon Radio but according to [this site](https://tremend.com/blog/embedded/change-your-bluetooth-address-of-your-linux-machine/), CSR produces 70% or more of bluetooth chips.  I can make the mac address change survive reboots or even change it permanently on chip.

Run `hcitool dev` to get the list of addresses
Then run `bccmd -d hci0 psset -s 0 bdaddr 0x44 0x00 0x66 0x55 0x33 0x00 0x22 0x11` replacing hci0 with the device number you want to change.  Use -s 1 to -s 3 to store the address on chip. (highest number has priority)

MattMorganMDP, 2024-04-20:
Also wondering what other troubleshooting can be done when unable to connect more than 5 controllers. Have multiple bluetooth receivers, with different addresses, and all controllers paired. Running of a Rapsberry Pi 5. Can't seem to get more than 5 connections going, though.

KilianSSL, 2024-05-13:
For me, neither bdaddr nor bccmd seem to be installed, despite i have BlueZ installed. Does anyone know why or can recommend another tool to change the mac address?

kurainisei, 2024-09-19:
Same here, I have no idea how to solved this, I tried many things but bdaddr refuses to install (it gives me a corrupted file error on unzipping).
If anyone could be so good to just add some step to step instructions that would be lovely.

KilianSSL, 2024-09-19:
@kurainisei i managed to change the address in the end, i'll see if i can post something tonight.

mattdsteele, 2026-10-14:
I've found that using the [currently-recommended adapter](https://github.com/adangert/JoustMania#recommended-bluetooth-adapter) is a CSR-based chip that will often have duplicate MAC addresses (the two I purchased were identical).

Raspbian includes bluez, but several of the tools (including bdaddr) are no longer included in the distribution, and must be compiled from source. Here's what worked for me:

```bash
# required toolchains
sudo apt install build-essential libglib2.0-dev libdbus-1-dev libudev-dev libical-dev libreadline-dev git autoconf automake libtool

# compile bdaddr
git clone https://github.com/bluez/bluez.git && cd bluez
./bootstrap
./configure --enable-testing --enable-deprecated --disable-systemd --disable-udev --disable-obex --disable-cups --disable-manpages
make tools/bdaddr

# find adapter mac addresses, should have hci0 and hci1, e.g.
hciconfig -a

# change mac address of one device; I just adjusted the last digit
sudo tools/bdaddr -i hci0 -r 00:1A:7D:DA:71:0A
```

This particular device appears to actually persist MAC address rewrites across unplugs and reboots.