adangert/JoustMania #173: look into psmove api not installing correctly with setup.sh

adangert, 2018-08-27:
from an email comment: because the package downloaded with joustmania wouldnt let me make any changes to it. Also, the psmoveapi was missing hidapi files, and once I put them in myself, everything installed just fine!

geirskogul, 2018-08-30:
How exactly is this fixed?  Joustmania installs without errors, but won't run.  Trying to manually run it gives the following error:

File "/home/pi/JoustMania/piparty.py", line 1, in <module>
import psmove, pair
ModuleNotFoundError: No module named 'psmove"

adangert, 2018-08-30:
I'll try to look into this asap.

geirskogul, 2018-08-30:
Thanks!  I cloned hidapi and tried to do the same with an older branch of psmoveapi, but I'm unfamiliar enough with this to not be able to figure out what files would be missing where as alluded to in that email fix you posted a few days ago, so I'm stuck.

geirskogul, 2018-08-30:
I got it to work through a kludge.  I opened up setup.sh, and added this line after "cd psmoveapi":

git reset -- hard f42c650f066f6835ec1bf531c13ab6a90c401f62

This reset the version of psmoveapi to the Sep 17 2017 commit for the rest of the installation.

Now it works with that older version.  I wonder what happened to the latest version of psmoveapi?

adangert, 2018-08-30:
Do you know what the error is without the hard reset?

geirskogul, 2018-08-30:
During setup.sh install, it seems to progress along normally, and seems to complete without errors.  However, it's as I posted above, running JoustMania manually in a terminal gives this error:

File "/home/pi/JoustMania/piparty.py", line 1, in 
import psmove, pair
ModuleNotFoundError: No module named 'psmove"

And connecting the controllers via USB doesn't sync them, and there's no menu music so I'm assuming piparty isn't running at all and psmoveapi must not be installed I guess.

So maybe it's not installing the module at all due to something being renamed or similar in the latest commits of psmoveapi.  The version I used for the reset is one I randomly pulled from the 2nd page of commits, just to try to go back far enough.  I haven't tried walking the versions forward at all to see where exactly it breaks, but I guess that would be the next step to figure out what caused it.

VicBond007, 2018-09-02:
Was going to post this along with my workaround.
My install failed when running setup.sh, something to the effect of Unknown CMake command "swig_add_library".  I think psmove was generating it.  I didn't have the foresight to back up the log (I thought new installations would append, not overwrite it).

Turns out swig_add_library was added to Cmake 3.8.0.  The latest Raspbian (June 24 2018) lists the very outdated cmake 3.7.2 in its apt repo, so updating to the "latest" still isn't late enough for whatever "new" changes were made to the PS Move software.  

I had to manually install a newer version of cmake by running the following commands from Terminal:
sudo apt remove cmake
sudo apt purge --auto-remove cmake
wget https://cmake.org/files/v3.12/cmake-3.12.1.tar.gz
tar -xzvf cmake-3.12.1.tar.gz
cd cmake-3.12.1/
./bootstrap
make -j4
sudo make install

This took FOREVER to compile, but after reboot, I could then run setup.sh properly.  The game appears to run fine, but I haven't tested it "for real" yet.  Just some basic controller pairing and shaking.

Full disclosure, I have zero clue what I'm doing.  There may be a much better way to accomplish this.

geirskogul, 2018-09-02:
An update on my workaround, I've walked up the psmoveapi commits and the newest one that works flawlessly for me is aed4e220cc80e9dd6c57605eb727daf22d6bae5d (Feb21), if anyone wants to kludge it that way until raspbian updates.

Newer commits mostly work, but for some reason the psmoveapi version from April 5, while it works in game, causes menu music to not play.  Weird.

adangert, 2018-09-04:
Installed the new version of cmake, things seem to be working now with the new psmoveapi.

closed 2018-09-04 by adangert