adangert/JoustMania #182: Audio stuttering when files initially loaded

martinbogo, 2019-01-15:
On my Pi 3 B+, with a high-performance SanDisk SD card ( 128GB, formatted EXT4 ) 

When initially loading audio files, there is a "stuttering" during playback, where the audio is rendered glitchy the first time it is played.  The second time it is played, there might be some caching involved because playback is quite smooth.   

I think this is due to the default IO scheduler currently used by default in Raspbian Lite.  I believe the default scheduler for the SD IO is CFQ, and should probably be set to the noop scheduler when booted.

/etc/udev/rules.d/60-mmc-scheduler.rules

# set noop scheduler for non-rotating disks
ACTION=="add|change", KERNEL=="sd[a-z]", ATTR{queue/rotational}=="0", ATTR{queue/scheduler}="noop"
# set noop scheduler for multimedia cards (SD-cards)
ACTION=="add|change", KERNEL=="mmc*", ATTR{queue/rotational}=="0", ATTR{queue/scheduler}="noop"

adangert, 2019-01-15:
are you using mp3s or wavs?

martinbogo, 2019-01-15:
Mostly Mp3's encoded 128kbit constant bitrate instead of VBR.

adangert, 2019-01-15:
Did the update fix it for your playback?

martinbogo, 2019-01-15:
Unfortunately no.  But when I switched IO schedulers to 'noop' it did fix the issue.  I think its just that the new default IO scheduler in Raspbian Lite is not optimal when there's a lot of file IO going on after initial boot and initial loading of the files before they get cached by the kernel.

adangert, 2019-01-15:
ok I'll see about changing that to be the default, I use the desktop version of raspian so perhaps that's why I'm not seeing the issue.