adangert/JoustMania #198: when joustmania starts have it notify the player if there is an update available.

adangert, 2019-04-03:
So they can update it on their own time, perhaps have a button to update joustmania when it boots up too.

adangert, 2019-04-03:
hold in start and select for three seconds to update.

ChainmailBuddha, 2019-04-03:
How would this work out though for those who leave it in AP mode most of the time?

adangert, 2019-04-03:
I think that it would only check for updates when connected to the internet. so you would have to switch it over or plug in an ethernet cord.

SkyroOryks, 2019-04-03:
This a good idea, since the git pull takes a little bit of a process. And like me one forgets to do it.

ChainmailBuddha, 2019-04-03:
Perhaps have it keep a record of the last time it attempted to update, and have an audio prompt on startup if it has not checked for an update in a certain period of time, such as 30 days or so.
Just as a reminder for those who don't always connect it to a network.

adangert, 2019-04-24:
Joustmania now will check for updates on startup, and update the program if it is a small update, it will prompt the user to update on a larger update.

closed 2019-04-24 by adangert