adangert/JoustMania #200: Increased sensitivity over time, AKA Sudden Death mode

ChainmailBuddha, 2019-04-03:
As some game modes can go on for longer periods of time, I would like to suggest an option (preferably a toggle of some sort) to enable sudden death mode.

After a fixed amount of time (possibly different for different game modes) have the sensitivity of the remaining/active players slowly ramp up from the base line. This could be accompanied by a corresponding increase in the music tempo, and audio alert, and or vibration feedback in the controllers.

Alternately, rather than time limited, have the sensitivity go up in correlation with the number of players already eliminated in that particular game. This method would have a greater impact on games with more players, where more people may be waiting to play, and less of an impact with smaller groups.

Yet another option would be to allow eliminated players to affect the tempo by holding the trigger on their controllers etc.

adangert, 2019-04-03:
something similar is implemented already, the fewer players there are the more likely it will go to and stay in the low sensitivity mode which encourages players to fight rather than move around slowly. more of these suggestions would probably help as well.

ChainmailBuddha, 2019-04-03:
Yes, but currently even the higher sensitivity is not all that sensitive, and in my experience the current implementation is not really noticeable and doesn't seem to impact play much.