adangert/JoustMania #205: Broken with new raspbian stretch?

MrAbodi, 2019-05-02:
Has the lastest raspbian broke this?
I’ve tried three times now to get it working and it just wont work, I’ve had no issues In The past.

Could someone share with me a working pi3 image? I‘ve got a relatively short timeframe to get a working version up and going for a big event.

adangert, 2019-05-02:
Strange, I tested this out with the new raspberry pi image a couple days ago, what are the errors you are getting? it might be that a third party service is failing. I'll retry it again tonight, and see if I can get a working image over to you.

MrAbodi, 2019-05-02:
I just tried again and this time I ended up with a JoustMania folder and a Joustmania folder, all the files are in Joustmania but of cause they want to be in the JoustMania folder.

I’m so confused why this is happening

MrAbodi, 2019-05-02:
Enable ap script doesn’t work for sure.

And once I moved all the house files to the correct folder and specifically ran joust.sh the game worked bit it doesn’t auto run on boot

adangert, 2019-05-02:
This is the file that is trying to run on boot: https://github.com/adangert/JoustMania/blob/master/conf/supervisor/conf.d/joust.conf

in setup.sh the line that enables this is:

#installs custom supervisor script for running joustmania on startup
sudo cp -r /home/pi/JoustMania/conf/supervisor/ /etc/

So yeah I would remove the Joustmania folder with the lower case m and make sure that the /etc/ has the /supervisor/conf.d/joust.conf file.

adangert, 2019-05-02:
if anything running sudo setup.sh from within the JoustMania folder should fix most problems.

MrAbodi, 2019-05-02:
Copied those etc files across and it did indeed get the auto start working.
I sincerely thank you for your help.
I have one final question if I may.

How can I ensure the sound comes from my USB sound card rather than the hdmi or audio jack if I needed too? I’m just not sure with the audio set we will be using wether sound from hdmi will be possible

adangert, 2019-05-02:
In the top right corner of the screen you hopefully can right click the sound icon and select the source of the output. Although I would say the best bet would be to just use the audio jack. I haven't tested it much with an audio card.

Glad everything worked out :)

closed 2019-05-02 by adangert

MrAbodi, 2019-05-08:
Hey Aaron, Thanks a bunch for the last minute tech support on
Thursday/Friday.

I wasn't able to get the Pi to act as an AP in order to admin from my
phone, but otherwise I was able to get it happening and we had 11 players
going at once, with maybe 35 people either playing or watching. It was a
big success and that is thanks to you. so Bravo sir.

I'll be sure to back up this mostly working version before trying to update
next time ;)

Many thanks,
Samuel