adangert/JoustMania #208: Feature Suggestion - Battle royal gametype

MrAbodi, 2019-05-08:
Really simple this one, The Idea is that you would have 2 - 3 concentric circles, and as players dwindle it would tell you to move into the smaller circle.  i would think this would make for tight end game bouts. 

benefit would be the joust organiser would be able to set the circle size to whatever they would prefer seeing as the game itself isn't tracking it.

adangert, 2019-05-08:
I like it, I think it would have to be one of the larger game types, like 3 circles, with 4 players each, so 12 minimum

ChainmailBuddha, 2019-05-08:
During my last big group event we did an alternative to this when we got down to a small group of players (and if they were not engaging, or being too defensive). Simply the players who had been eliminated formed a circle with our arms out around the remaining players that slowly closed and restricted the play area.

Turned out to be a very effective way to force an end to a match that was going on too long. :)

MrAbodi, 2019-05-09:
@ChainmailBuddha yeah that works to when you have the open space and people to surround the play area, that hasn't been a situation i've found myself in, which made me think of this idea. :)

ChainmailBuddha, 2019-05-09:
@MrAbodi I dunno, we did this in a meeting room on a cruise ship that was maybe 10-15' square, with about a dozen people playing. So I don't think it takes too much in the way of players or space.

Not to knock your idea at all. Just providing an alternative that has been tried and tested :P