adangert/JoustMania #209: AP not working since last major update

ChainmailBuddha, 2019-05-10:
When I run the enable_ap.sh script for the AP, the script runs, Pi reboots, wifi is disabled, and everything seems to be fine and working like it always did. However the SSID does not appear on other devices.

NB: Additionally, when either the enable or disable scripts are run, the first thing that appears is an illegal number error. However this has been occurring for a while now with no other apparent effect on either script.

adangert, 2019-05-10:
the illegal number issue should be taken care of, the AP still seems to be broken

adangert, 2019-05-13:
it's somewhat fixed for now, you can enable the AP then try going to http://10.3.141.1 in your browser, I've tried it with my macbook, phone seems to not be able to connect for me though

closed 2019-05-13 by adangert