adangert/JoustMania #215: Newest Raspbian breaks psmove controllers from connecting.

adangert, 2019-05-23:
For anyone that is installing the latest raspbian and find that controllers are not connecting, try flashing this image instead, it is known to work:

https://drive.google.com/file/d/1zffvRIWWV2TGGKgmj0qCp_EwlnU6-Y49/view?usp=sharing

Another issue was opened up in the psmoveapi repository:
https://github.com/thp/psmoveapi/issues/385

adangert, 2019-05-23:
the raspberrypi-kernel (1.20190517-1) over (1.20190401-1) was causing this issue. Downgrading the kernel via 

sudo rpi-update 00bb2894c0a4349464c19ba62f62b8d9e17a1a9e

solves the issue for now, and has been added to the setup.sh file. Although this is a temp fix, as new kernels will be released in the future.

closed 2019-05-23 by adangert

MWDelaney, 2019-07-04:
> sudo rpi-update 00bb2894c0a4349464c19ba62f62b8d9e17a1a9e

I had to run this today on a fresh install. Not why the updated `setup.sh` didn't help here but this appears to still be necessary on Stretch :(

adangert, 2019-07-04:
Yeah that's in the setup.sh script, you have to press y though.