adangert/JoustMania #241: Feature Request - Optional "Force Game Start" function

Zanial, 2019-11-01:
Add an ability for players that have already activated their controllers to force a game to start, even if not all connected controllers are activated for the selected game.

Use Case:
Example: 12 controllers connected, Free For All game selected, 7 players have activated their controllers by pressing A.  No other players are available, other controllers must be disabled which is time consuming and requires knowledge of how to do so.  If a couple players could hold the trigger or some other button (maybe A again?) at the same time to cause the game to activate without the other connected controllers it allows more players to join in the next game easier.

ChainmailBuddha, 2019-11-01:
This would also be useful in the case where controllers have been disconnected, but JM has not recognized this (typically if controllers are disconnected after one or more players have readied themselves), which prevents the next game from starting automatically.

Games can be force started from the web interface, but that is not always available in all configurations.

Request seconded :)

adangert, 2019-11-01:
Maybe it could be if all current players hold the trigger buttons then it would Auto start

Zanial, 2019-11-01:
The difficulty with requiring _all_ players to hold the trigger is it requires them all to know it needs to be done, slowing the speed at which the games are normally played through.  Someone then has to be on hand to inform everyone of how to start the next game if there aren't max players.  I think it would also mean if a player readied themselves and their controller disconnected, you're still stuck.

In the original Johan Sebastian Joust it was any two controllers which seems to work well.  Generally in a setup when you have lots of people cycling in or out there's a few consistent players that can avoid stagnating a game by getting it going even with less than maximum players, otherwise people lose interest and move on.

ChainmailBuddha, 2019-11-01:
What about the option of having two or more controllers holding the triggers initiating a countdown for game start? This could also allow a window for additional players to join if they want, but still moving the game along. Somethings along the lines of: (2+ triggers held for 3 seconds) "The next game will start in 10 seconds. Please press your trigger now to join". <10 second countdown punctuated by beeps, with final three having the current stoplight colors>

adangert, 2020-01-27:
I've added the ability to force start games from the admin mode, by holding down the trigger button for two seconds the game will start with all the players that have pressed their trigger buttons. This should also allow for turning off all of the unused controllers during that game as well.

closed 2020-01-27 by adangert