adangert/JoustMania #242: Strike sounds are so loud the music and voice comments are hard to hear could the mix be adjusted via the web inteface?

TheMiniBar, 2019-12-15:


TheMiniBar, 2019-12-15:
Strike sounds are so loud or some sound systems that the music and voice comments are hard to hear. Being able to adjust these three types of audio would be great

adangert, 2019-12-15:
what are strike sounds?

TheMiniBar, 2019-12-16:
Sorry, my name for the sound the Pi plays when a PS Move controller is hit.

adangert, 2019-12-16:
I might be able to turn down that volume, you can do it manually via changing https://github.com/adangert/JoustMania/blob/master/audio/Joust/sounds/Explosion34.wav right now

TheMiniBar, 2019-12-16:
Ah cool. I was wondering whether. Voice instructions and could be put more on (for example) the right side of a stereo channel, and music on the other side (eg left) and then some sound systems with balance control could easily adjust the mix.  Just a thought. 

I might try toning down the volume of the explosion.wav if I can find suitable software. Individually people know when they are “hit” generally anyway. 

Thanks for all your efforts. 👍😀👍

TheMiniBar, 2020-01-26:
I can't upload WAV files here, what's the best way to send you the modified Explosion sounds?

TheMiniBar, 2020-01-26:
Just noticed after copying the modified explosion sounds that Joust announces “Joust Updated” (or similar) every time it boots up. Is there any implications from that message? (Eg my file gets overwritten after reboot, or I won’t get updates in future?). @adangert

adangert, 2020-01-27:
Oh you can make a pull request in another branch to upload the files. Also the updater just checks if files have been changed to do a pull request, so you should be fine, it will still update correctly. You can add the files to the .gitignore file to stop it checking those files.

adangert, 2020-10-01:
Lowered the volume of Explosion34.wav in the latest commit.

closed 2020-10-01 by adangert