adangert/JoustMania #243: The JOUSTMANIA SSID/web interface seems intermittent.

TheMiniBar, 2019-12-15:
Despite enabling the local JOUSTMANIA Wifi hotspot it’s rather hit and miss as to whether it is available. 
Am I missing something?

adangert, 2019-12-15:
are there specific issues you are having? you can also try logging into the pi, and going to joust.mania in the browser to check if its working.

ChainmailBuddha, 2019-12-16:
@TheMiniBar I was recently seeing an issue where the hotspot would come up about 50/50 per reboot of the Pi. Does that sound like the issue that you are experiencing?

TheMiniBar, 2019-12-16:
Exactly @ChainmailBuddha 
We’ve been playing with no screen but I’ll test more at home. 
The SSID for the local JOUSTMANIA hotspot just doesn’t appear available even though it’s a known network (and not a hidden SSID), around half the time. Sometimes a reboot makes it appear. Could be a hardware issue potentially just on my Pi I guess but we are seeing exactly the same symptoms @ChainmailBuddha described.

ChainmailBuddha, 2019-12-16:
@TheMiniBar Just to gather some more data, are you on a Pi3 or 4? If you are on a 4, what are you using for power, and do you happen to get the lightning bolt symbol when you do get a chance to connect it to a screen?

TheMiniBar, 2019-12-16:
A Pi 3b

TheMiniBar, 2019-12-16:
Using a pi branded adapter. Doesn’t ever have trouble connecting to our Wi-Fi at home. Haven’t ever seen the lightning symbol but have had temperature warnings. Bought a case with heat sinks and a tiny fan and now the temperature is always good

ChainmailBuddha, 2019-12-16:
Ok, thank you for the update. So @adangert we now have this issue on both the 3 and 4, so it should not be hardware or power issue as possibly suspected before. I believe this issue began after the last round of updates to fix the AP, so there may be a bug there.

adangert, 2019-12-16:
Yeah I'll have to take a look at it, the web interface has been notoriously difficult to debug and get things right, it's probably some software issue with some service not starting when it needs to. The best place to look would probably be in /var/log/messages to see if anything is amiss.

TheMiniBar, 2019-12-16:
Would it help to test if I disable the local AP and re-enable to see, if it appears every time if switched off and on again?

adangert, 2019-12-16:
yeah go for it.

TheMiniBar, 2019-12-17:
Ironically tonight the JOUSTMANIA AP has worked 3 out of 3 reboots with no settings changes. Total guess but is it possible that the the channel is fixed or limited and some external factor makes the AP seem intermittent? (Total guess)

adangert, 2019-12-17:
I believe so, you can see the settings here: https://github.com/adangert/JoustMania/blob/master/apfiles/hostapd.conf

When the script is run it replaces these files with the ones in the system.

TheMiniBar, 2019-12-17:
If the channel is fixed in the config file in a densely populated area maybe that would cause problems. In the UK our home wifi routers generally autoselect a channel based on availability after checking what else is broadcasting. Could a clash make no JOUSTMANIA appear, or should it still appear if the device connecting is right by the pi? I would have assumed so, so perhaps my channel clash theory is just wishful thinking?

adangert, 2019-12-17:
That could be it, also changing some settings back to what they were previously would probably be the best plan of attack: https://github.com/adangert/JoustMania/commits/master/apfiles/hostapd.conf

adangert, 2020-06-25:
Yeah I was just testing it out and couldn't see an AP for the Pi on the latest attempt.

adangert, 2020-09-13:
I recently did a sudo apt-get update/upgrade on the pi and was testing specifically for whether or not the AP was showing up or not. In all five times that I rebooted/enabled and disabled the AP the JOUSTMANIA ssid showed up. Perhaps some pi upgrade fixed this issue. I'm going to close this for now, but if it reoccurs for anyone, please reopen this so we can investigate further.

closed 2020-09-13 by adangert

TheMiniBar, 2021-10-17:
I'm still getting this issue. Now on a 4b