adangert/JoustMania #247: Fresh install now says “broken packets”

TheMiniBar, 2020-01-25:
After our Joust Install announced a major update was available, Joust Mania broke.
Long story short, even flashing a clean OS Pi SD card and downloading from github and running ./setup.sh I still can’t get a working Joust Mania install. 
We were due to play this afternoon and I warned people I was having to do a clean install to fix things, but that didn’t work, and we called it off. 
[image]

adangert, 2020-01-25:
This is strange, I just flashed a new pi an hour ago, and it worked for me. I used the Raspbian Buster with desktop option

TheMiniBar, 2020-01-26:
I had two SD cards all setup for Joust, and both give the same error,  then I made a third from scratch using the same image and it’s done the same thing. On the new image I did accept a long download of updates for the pi before pulling in Joust from GitHub. 

Not sure what else I can do now as I have three SD cards all with a non-working Joust. I suppose the one thing left is to not download updates for the Pi. We were due to reconvene today, but I’ll call that off. 

I guess there could be a hardware fault of some kind on the Pi, but feels unlikely. It’s working fine without any complaints for standard Pi stuff. Just ./setup.sh failing consistently every time on all three cards.

Suspect all would have been well if I’d not tried to accept the large update on my first two cards via holding select and start

TheMiniBar, 2020-01-26:
I’ve been using Raspbian Buster Full version from the start and still used it on the new one yesterday to try and fix, but I’ll try the desktop version without the recommended software and will avoid updates before mirroring GitHub

TheMiniBar, 2020-01-26:
Are you on Pi 3b or 4? 
I’m using a 3b. 
Will try on a borrowed 4 to eliminate the possibility of hardware issues.

TheMiniBar, 2020-01-26:
All fixed on our Pi 3b
This command line before running ./setup.sh fixed things....

sudo apt-get dist-upgrade

closed 2020-01-26 by TheMiniBar

TheMiniBar, 2020-01-27:
Well it’s nothing to do with getting an update of either JoustMania or anything Pi OS that caused the problem on my 3b

Got a brand new 4gb Pi 4b delivered today with Raspian Desktop preloaded and afternoon doing the GitHub Mirror and running ./setup.sh I got the same broken packets error message and couldn’t get JoustMania installed. 

Running sudo apt-get dist-upgrade pulled a load of Raspian files in (way out of my areas of Pi knowledge) and after that JM setup.sh has worked and we are up and running on a new Pi 4b with our 3b now becoming our backup system in case of future failures.

adangert, 2020-01-27:
Did you do the initial raspberry pi update (following the instructions when it's first booted for the first time?) or do you skip those steps?

TheMiniBar, 2020-01-27:
With the Pi4b I can’t remember for sure (sorry). I definitely didn’t decline updates, but I don’t remember it going through a lengthy update process before it was up and running. Part of me wonders if the Auto updates failed for some reason. 
It’s weird that setup.sh failed even from my Reimaged Pi. It was an image I’d downloaded months ago, so perhaps the image was out of date for the new version of JoustMania? 

Separately, I’m just practising making images of a clean install of JoustMania for backing up to my computer and being able to swap between the 3b and 4b quickly if one failed. I’m resetting the Bluetooth connections on both Pi’s before saving the image so that which ever Pi is brought out to the save the day in future, it’ll be ready to just see the PS4 controllers via a lead and then we play....  

Thanks for all your efforts on JoustMania. It’s the highlight of our weekend getting friends together to play in our church hall. 

Bill