adangert/JoustMania #249: Add ability to force start with either all or some of the players as an option

adangert, 2020-01-27:
have this both in the webui, and on the admin menu.

ChainmailBuddha, 2020-02-20:
Admin mode force start does not appear to be working reliably. Intermittently I have had this command do the following:

- Work as intended

ChainmailBuddha, 2020-02-20:
Admin mode force start does not appear to be working reliably. Intermittently I have had this command do the following:

1. Work as intended

2. Start the readied controllers, but not let the game end when one controller is remaining. Remaining controller can be killed as normal.

3. Start all paired controllers as if they had been readied, but not let the game end when one controller is remaining. Remaining controller can be killed as normal.

NB: For the last two options this also seems to kill both Joust and the webgui server.

For what it is worth, the rest of the game seems to work normally as long as the force start option is never touched.

adangert, 2020-02-27:
updated in latest commit, also tested with turning off controllers in the middle of gameplay that were not added during the forced start (this will allow you to remove controllers while others are playing)

TheMiniBar, 2020-02-27:
This is great! We've had some controller drop outs mid game lately or mid game quitters and this will make the game play flow much better with the game still being completable. 
Thanks Aaron

adangert, 2020-02-27:
Oh I haven't tested it for controllers that are actively apart of the game, just the ones that's were not included in the game due to the force start. I think the former would take some more code changes to be robust

adangert, 2020-06-25:
Updated with new option both in the admin controller section and the webui, added the random_team_size in the webui

closed 2020-06-25 by adangert