adangert/JoustMania #250: More than 2 USB Bluetooth Adapters on a Pi4b?

TheMiniBar, 2020-02-18:
My 3b lets me plug in a third Bluetooth adapter, reset the Bluetooth connections  and start pairing controllers. No problems. 

On the 4b I can’t get a third or fourth adapter to pair with the controllers. Only two adapters. Complete rebuild, GitHub clone, etc and still only 2 adapters usable. 

Anyone else experiences this?

adangert, 2020-02-18:
I have 4 adapters working on my 4b, have you tried with a different combination of those 4? Also I think there was a problem where if two adapters had the same MAC address there were problems.

TheMiniBar, 2020-03-14:
I resolved this. Bzarrely one of the four brand new Trendet adapters had a hardware fault. All different MAC addresses. It presented different missing bluetooth adapters depending on which socket the faulty adapter was placed in.  
Not a JoustMania issue at all but hardware. 

Now running on the 4b with four adapters And 18 controllers.

closed 2020-03-14 by TheMiniBar