adangert/JoustMania #252: Suggestion: Secret Pi reboot triggered by controller?

TheMiniBar, 2020-03-09:
Hi there. Apologies if this isn't the right place for a Pi version feature requests. 
We are playing with between 12 and 18 controllers but lately hit problems with controller drop-outs and a wifi AP crash out where the web-AP stops working and so reboot unavailable without a screen and mouse. A mid-season clean re-boot (rather than power-disconnect) does tend to resolve things. 

Is it possible to build in a "cheat code" that would allow a controller to rebooy the Pi or at least stop and restart the Joustmania software?

Alternatively I perhaps should carry a small screen and mouse with me for rebooting or re-pairing Bluetooth controllers but we are really hoping for an off-grid outdoor setup for the summer.

adangert, 2020-03-09:
Do all the controllers disconnect when this happens?

TheMiniBar, 2020-03-09:
No - just seemingly random ones. I can take the disconnected controller out, introduce another and then later another random one disconnects. It does happen it's often mid game.

adangert, 2020-03-09:
hmmm, I should check what happens when one controller moves too far away from the play area, or some sort of signal disruption, maybe that causes a disconnect.

TheMiniBar, 2020-03-09:
Yes we have only played in a fairly small Hall in two different locations but with controller drop out. 

I have carefully paired six controllers per Bluetooth dongle and played with seen all controllers responding well at home in the past but lately even lined up in a row right near the Pi I can see occasional lights going off and then a few seconds later them coming back on. 

How are you using 4 USB dongles and able to pair controllers using a lead? Do you introduce each dongle in turn selecting which controllers to pair with that dongle and then repeating again with a different dongle and another set of controllers?

Or do you use a USB hub for pairing with all four dongles together?

TheMiniBar, 2020-03-09:
I should say that in my experience that the controller light goes off, the red power led remains steady and the joust.mania Page shows the controller connected, but when it happens midday game the controller is unresponsive. For example in Fight Club if a controllers light goes out without explanation their controller is sometimes still selected to fight and they are of course undefeatable with a defunct controller and so game play never can pass to a valid controller. 

I wonder if it's actually a problem with a dongle. So hard to tell.

adangert, 2020-03-10:
yeah for me I use a dongle with all of the bluetooth controllers plugged in, Joustmania will split up the controllers to the bluetooth dongles that have the least amount of controllers connected, Also make sure that your internal bluetooth is deactivated since there was a bug in the setup script that got patched recently.

TheMiniBar, 2020-03-10:
Thanks Aaron. I will invest in a USB hub then and try that. 
I rolled back to a previous version of Joustmania and continued to get drop outs but at the end found two controllers connected to the internal Bluetooth. 

You don't use the internal Bluetooth in your 4b then? Just 4 USB Bluetooth's via one USB hub connected to the Pi? 

Will just about any 4 port hub work with a Pi?

adangert, 2020-03-10:
Well yeah the pi has 4 USB slots, so I plug three into the pi, and one into the dongle so to pair the controllers, after pairing I remove the USB hub and plug all four dongles into the pi itself. Also yeah the internal Bluetooth is very unreliable and slow, and previously the --disable_internal_bt command wasn't working properly.

closed 2020-03-10 by adangert

TheMiniBar, 2020-03-10:
Ok thanks.