adangert/JoustMania #253: Force starting Non-Stop Joust crashes the AP and JoustMania

TheMiniBar, 2020-03-14:
I get  "Not enough players" audio message and have to reboot.
12 controllers show as connected but if I force start from the AP it crashes the AP, all controller lights go out and my only option appears to be to reboot

I'm running 503d6f3 but am being dim and can't see where I check  what the latest release version is. 
Apologies if this is already resolved in the latest version.

TheMiniBar, 2020-03-14:
Actually.  Don’t waste any time on this until I’ve reflashed a completely vanilla 4b Pi Raspian install and done a new git clone of JoustMania l. It could be I’ve got some file corruption from a previous session where we had to power off rather than shut down cleanly.

adangert, 2020-03-14:
thanks for reporting, I'll give it a try too

adangert, 2020-04-22:
I tried for starting with 4 controllers and it seemed to work if you have this issue again let me know.

closed 2020-04-22 by adangert