adangert/JoustMania #256: get audio to work out of headphones when plugged in

adangert, 2020-05-30:
currently sudo aplay <wavfile> will only play out of the HDMI audio when plugged in to a monitor, this is a new change that was not there previously (something in the new raspbian version changed this it seems). Since joust runs in sudo mode this is how audio will be output for now. I've made an issue here to see if it is resolved at all

https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=275609&p=1669748#p1669748

adangert, 2020-06-09:
Likely will not be fixed, will just have to use hdmi audio when plugged in.

closed 2020-06-09 by adangert

TheMiniBar, 2020-06-09:
So I guess this means that the raspi-config no longer has the automatic  audio output option in stead of always headphones or always HDMI?

I must confess I've not powered up my Pi since lock down began. We are missing playing Joustmania 😌

Bill

adangert, 2020-06-09:
I'm not sure I haven't looked at the raspi-config options

ChainmailBuddha, 2022-02-19:
Accidentally found the solution to this, as long as you are ok forcing audio to the analog jack. Which in my case is exactly what I wanted. Note this solution is for Raspbian Buster, and may not be relevant or apply to the new dedicated Pi OS
The answer is here: https://raspberrypi.stackexchange.com/questions/112954/how-to-restore-audio-output-after-updating-raspbian-buster-on-pi4

But the tl;dr important bit is:
Add the following lines, either to your /etc/asound.conf file (global configuration) or to your ~/.asoundrc file (per session configuration), as you prefer/need.

defaults.pcm.card 1
defaults.ctl.card 1

if you already made similar settings, change them or add them to the config file. If the file does not exist, create it.