adangert/JoustMania #263: Controllers "paired" but not lighting up?

noahlemen, 2021-09-24:
I am running a Raspberry Pi 4 (4GB model) with the suggested bluetooth dongle and USB hub. My 7 move controllers appear to pair, but never actually light up white. The web UI shows all 7 controllers as paired and active, but seems to perpetually show low battery on all of em (even after charging overnight). I tried running `reset_bluetooth_connections.sh` to no avail. The setup script did not encounter any errors.

Any advice for troubleshooting this?

adangert, 2021-09-24:
the newer ps4 controllers (micro usb not mini), should turn white when you plug them in, then after that when you press the sync button, they should turn orange once connected to the pi. 

sudo ./reset_bluetooth_connections.sh should reset all the bluetooth connections so you will have to repair the controllers. I would advise going one controller at a time.

Also you can kill the joustmania process using sudo ./kill_processes.sh

then manually start joustmania with sudo ./joust.sh
then you will be able to see the output in the terminal. You should be able to debug what's going on there. The new move controllers (micro usb) might take a couple of times of getting it plugged in and pressing the sync button for them to work. I hope this helps

noahlemen, 2021-09-27:
Thanks for confirming that! 

Still not seeing a clear culprit. I'm getting a mysterious unattributed `Home directory not accessible: Permission denied` message in the output of `joust.sh`. Tried correcting ownership via `sudo chown -R $USER:$USER $HOME/`, but still see the message. Is ownership expected to be configured differently?

Only other potentially-suspicious output I'm seeing is `[PSMOVE WARNING] Magnetometer in <<MAC ADDRESS>> not yet calibrated.` for all of the controllers. 

FWIW, only one of the 7 controllers is one of the newer ones with micro USB.

adangert, 2021-09-27:
joust.sh should be running with sudo, so the permission error is a bit strange. I would also make sure that your dongle is flashing blue as well.
If all else fails, I would just do a clean install of the raspberry pi os and do the three commands for installation. You should hear music from the HDMI port if everything is working correctly on boot.

adangert, 2021-09-27:
if you can hear music it means the game is running, if you can't pair controllers then that might mean that something went wrong with the psmove api installation.

You could also do a clean install without disabling the internal bluetooth to see if the controllers might connect then. It could possibly be a dongle issue in that case.

noahlemen, 2021-09-27:
Sounds good. Clean install seems worth trying, I'll see if that makes any difference.

adangert, 2021-09-27:
yeah I could compare the setup.log with mine if it doesn't work to see if there were any problems when it installed.

Also for reference I use the Raspberry Pi OS with desktop option on the website, and on first boot, I do the initial software update that is prompted by the OS.

TheMiniBar, 2021-09-27:
Great thank you.
I've chosen Pi OS Full. Perhaps I hadn't chosen that last time. 
Going to rebuild tonight. :)

TheMiniBar, 2021-09-27:
I've seen those magneto meter errors when trying to get the windows version running. What OS are you installing on?

adangert, 2021-09-27:
yeah the magneto meter errors I think happen on both the pi and windows version. Basically since we don't use the magnometer it throws those errors I believe.

closed 2021-10-08 by adangert

TheMiniBar, 2021-10-21:
Did you try resetting one of the controllers that wouldn't pair? There is a reset button and instructions online as to how to do this. 

I had one early install where I had similar problems but I ended up starting with a fresh install of the most comprehensive Pi OS desktop, then doing the clone GitHub for Joustmania and then taking an image of that SD card so I could easily refresh everything if I hit problems again. After that, if I get problems I just refresh the SD card with my backed up image. 

A rebuild from scratch sorted it for me. I never worked out why I couldnt pair all my controllers

adangert, 2021-10-22:
I have had bad Bluetooth dongles before, there's a chance you might have just gotten an unlucky one,

I would try reloading the os again, not plugging in the controller and leaving off the --disable_internal_bt line,
Then see if the controllers just pair with the raspberry pis internal Bluetooth