adangert/JoustMania #264: Pi 4 fresh install and sudo./joust.sh gives error - we can't play :(

TheMiniBar, 2021-09-26:
Fresh raspbian image on Pi 4 
Github clone done. 
Bluetooth internal disabled. 
But no Joustmania
Sudo./joust.sh from the Joustmania directory gives  "./joust.sh line 10: /home/pi/JoustMania/venv/bin/python3: No such file or directory"

This fresh build was produced 25/9/2021

TheMiniBar, 2021-09-26:
joust,sh contains this:
#!/bin/bash

if [ $UID -ne 0 ]; then
  echo "Not root. Using sudo."
  exec sudo $0
fi

export HOME="/home/pi/JoustMania"
export PYTHONPATH="/home/pi/psmoveapi/build/"
exec /home/pi/JoustMania/venv/bin/python3 /home/pi/JoustMania/piparty.py

TheMiniBar, 2021-09-26:
@adangert any ideas? Trying to get sorted for a Youth Group event. I have a Pi3b with an old install that won't update 'Update failed' message, but would be more comfortable on a fresh updated Pi4b ideally. Apologies for being a bit of a Pi dunce

TheMiniBar, 2021-09-26:
Totally guessing, but the build output has references to what looks like a structure that is probably on your Pi...
C:\Users\Aaron\Documents\GitHub\JoustMania\psmoveapi\scripts\visualc>build_msvc.bat 2017
The system cannot find the path specified.
Applying libusb dynamic CRT patch...
C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/scripts/visualc/../..//scripts/visualc/libusb_dynamic_crt.patch:10: trailing whitespace.
      <RuntimeLibrary>MultiThreadedDebugDLL</RuntimeLibrary>
C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/scripts/visualc/../..//scripts/visualc/libusb_dynamic_crt.patch:19: trailing whitespace.
      <RuntimeLibrary>MultiThreadedDLL</RuntimeLibrary>
C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/scripts/visualc/../..//scripts/visualc/libusb_dynamic_crt.patch:32: trailing whitespace.
      <RuntimeLibrary>MultiThreadedDebugDLL</RuntimeLibrary>
C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/scripts/visualc/../..//scripts/visualc/libusb_dynamic_crt.patch:41: trailing whitespace.
      <RuntimeLibrary>MultiThreadedDLL</RuntimeLibrary>
warning: 4 lines add whitespace errors.

adangert, 2021-09-27:
That looks like it was not able to set up the virtual environment correctly, I did a fresh install a couple of days ago and it worked. I would potentially try doing a full install again and looking at setup.log for any errors

adangert, 2021-09-27:
If you run /home/pi/JoustMania/venv/bin/python3 you should get a python window to open up of everything was done correctly

TheMiniBar, 2021-09-27:
I'm going to make sure I chose the correct PiOS Option and do a full install then will do the python check you've suggested if I hit errors and will take a look at setup.log. Thanks Aaron

TheMiniBar, 2021-09-28:
Think I'm sorted. Thanks for your help! 😁

closed 2021-10-08 by adangert