adangert/JoustMania #274: Software not starting on device startup

ipodholder385, 2022-05-13:
I have a Raspberry pi 4 model b and a 32gb micro sd card and I'm trying to install the software using the script in the walkthrough. Once the install is finished I hear "install complete, rebooting device" The device reboots, but Joust does not start. I'm unable to see if there are errors during installation as the command prompt goes by too fast and I see the Joust folder in the Home menu. But after doing some digging, I can't find where to start the software. I have flashed the sd card three times with the newest pi software. 
(I'm also extreme novice at this whole thing)

ipodholder385, 2022-05-13:
I do get an error message, 
joust@raspberrypi:~ $ cd JoustMania
joust@raspberrypi:~/JoustMania $ sudo ./joust.sh
/home/pi/JoustMania/venv/bin/python3: can't open file '/home/pi/JoustMania/piparty.py': [Errno 2] No such file or directory

I'm using the OS that came out on 2022/04/04

adangert, 2022-05-13:
Hi, it looks like you are using user joust rather than the default user pi. Please reinstall the OS and use username pi when installing. That should solve your problems.

closed 2022-05-13 by adangert

ipodholder385, 2022-05-19:
Just to confirm, the default user is pi . What about the password? can that be anything?

adangert, 2022-05-19:
the default password is raspberry so that might be better to use.

ipodholder385, 2022-05-19:
[image]

And just to confirm, this is the os that should be installed on the machine. The one that Raspberry Pi Imager v1.7.2 is defaulting to.

adangert, 2022-05-30:
Yes that should be the one that works. (or 64 bit)