adangert/JoustMania #276: Home Directory error

ipodholder385, 2022-05-15:
When I run the software (using the default user pi) I get the following error after typing in sudo ./joust.sh

pygame 2.1.2 (SDL 2.0.14, Python 3.9.2)
Hello from the pygame community. https://www.pygame.org/contribute.html
linux
Home directory not accessible: Permission denied
Home directory not accessible: Permission denied
Home directory not accessible: Permission denied
we found an exception when loading the settings!
 * Serving Flask app 'webui' (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
Address already in use
Port 80 is in use by another program. Either identify and stop that program, or start the server with a different port.
/home/pi/JoustMania
[]
1905341f501ac39f9d1a05f7d603a05f6b0a4a93
Home directory not accessible: Permission denied*

*this last line is repeated until I close the prompt.

I read through previous reports and reimaged the machine, but to no avail.

adangert, 2022-05-15:
Hi before running joustmania in the terminal (it should run automatically already) please run

./sudo kill_processes.sh

in the joustmania folder.

ipodholder385, 2022-05-16:
For some reason I'm getting a bash: ./sudo: No Such file or directory
error.

ipodholder385, 2022-05-16:
pi@raspberrypi:~ $ cd JoustMania
pi@raspberrypi:~/JoustMania $ ./sudo kill_processes.sh
bash: ./sudo: No such file or directory

adangert, 2022-05-16:
Sorry it's sudo ./kill_processes.sh

ipodholder385, 2022-05-16:
Traceback (most recent call last):
  File "/home/pi/JoustMania/venv/lib/python3.9/site-packages/dbus/bus.py", line 177, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/home/pi/JoustMania/venv/lib/python3.9/site-packages/dbus/bus.py", line 361, in get_name_owner
    return self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
  File "/home/pi/JoustMania/venv/lib/python3.9/site-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.bluez': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/JoustMania/piparty.py", line 1048, in <module>
    piparty = Menu()
  File "/home/pi/JoustMania/piparty.py", line 303, in __init__
    self.pair = pair.Pair()
  File "/home/pi/JoustMania/pair.py", line 17, in __init__
    self.hci_dict = jm_dbus.get_hci_dict()
  File "/home/pi/JoustMania/jm_dbus.py", line 14, in get_hci_dict
    proxy = get_root_proxy()
  File "/home/pi/JoustMania/jm_dbus.py", line 45, in get_root_proxy
    return BUS.get_object(ORG_BLUEZ, ORG_BLUEZ_PATH)
  File "/home/pi/JoustMania/venv/lib/python3.9/site-packages/dbus/bus.py", line 241, in get_object
    return self.ProxyObjectClass(self, bus_name, object_path,
  File "/home/pi/JoustMania/venv/lib/python3.9/site-packages/dbus/proxies.py", line 250, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/home/pi/JoustMania/venv/lib/python3.9/site-packages/dbus/bus.py", line 182, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/home/pi/JoustMania/venv/lib/python3.9/site-packages/dbus/bus.py", line 277, in start_service_by_name
    return (True, self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
  File "/home/pi/JoustMania/venv/lib/python3.9/site-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.bluez': timed out (service_start_timeout=25000ms)

adangert, 2022-05-17:
I'm not sure what this error is, I would look at the setup.log to see if there were any errors when installing. Or try re-installing the OS/game from scratch to see if that fixes the issue. If everything works right you shouldn't have  to run anything and the game should start at startup, and you should hear music out of the headphone jack port.

ipodholder385, 2022-05-17:
[setup.log](https://github.com/adangert/JoustMania/files/8704620/setup.log)
I've attached the log. I also changed sd card, imaging pc and even wifi and still unable to get Joust to boot.

ipodholder385, 2022-05-19:
Ok so I'm super dumb and it's been working this whole time. I never plugged the audio cable in correctly. That's totally on me.
One last question before you close this, can bluetooth 5.0 work with the ps move controllers from the ps3 and ps4? specifically this one
https://www.tp-link.com/us/home-networking/usb-adapter/ub500/

adangert, 2022-05-20:
Oh I'm not sure, but you can give it a try! The class 1 adapters listed on the main readme should theoretically go up to 100 meters.

closed 2022-05-20 by adangert