adangert/JoustMania #288: Open hardware design controller?

atatdotdot, 2022-08-31:
I just came across this game for the first time being run at the Greenbelt festival in the UK last weekend and loved it! I'd love to get hold of it for our youth group but the cost of the controllers is somewhat eye-watering if you want a decent number. I was thinking that it surely wouldn't be too hard to make one using an ESP32 – it seems as though it would just need an RGB led, an accelerometer and a Bluetooth connection.

I saw this issue from a number of years ago where it looks like some people were looking into the idea, but nothing much seems to have come of it https://github.com/adangert/JoustMania/issues/141 Opentilt appears to be dead. I've asked Dr Grant Pusey if he ever got round to publishing the code for his Microbit version https://twitter.com/atatdotdot/status/1564998893447823362 so we'll see where that goes.

adangert, 2022-09-06:
Thanks, it's always been an interest to add different controllers, we also thought that adding phone support would also help bring down the costs of controllers as well! https://github.com/adangert/JoustMania/issues/88

NickCiafardoni, 2023-06-06:
Is this something that is still being pursued at all? @adangert I was thinking about the phone/jackbox paradigm you mentioned in #88 - was also considering whether smart watches would be a good addition. I'm not a python expert, but I might be interested in trying to fiddle with that if you don't mind (who knows when there is time). How deeply is the app tied to the psmove controllers right now?

adangert, 2023-06-14:
Yeah I know that there were some people that were trying to work on implementations like that (One consideration was getting armbands for phones) since there was a likelihood that phones might get broken, or the sensitivity would have to be increased. I haven't thought of the smart watch idea, but that also sounds pretty neat as well!