adangert/JoustMania #289: ModuleNotFoundError: No module named 'psmove'

Mattman842, 2023-01-04:
I have tried to install JoustMania to a new RasberyPi 4 b 8Gb. Running on the latest version of Pi OS that the Raspberry Pi Imager recommends. I have tried running the rerunning Setup.sh multiple times. Also, different SD cards all within the recommendations. 

I have tried a whole ton of other things but I'm not confident with troubleshooting these, so they were most likely just shots in the dark. 

The error I get when trying to run Joust.sh

ModuleNotFoundError: No module named 'psmove' 

Just wondering if you've seen this error before and If I've just messed something up along the way.

adangert, 2023-01-04:
Yes that either means that the psmove api didn't install correctly, or that python is having trouble finding psmove. There are some example scripts within the psmove folder that should run if everything works correctly. I currently don't have a computer so I can't test from scratch at the moment, but should be able to next week

Mattman842, 2023-01-09:
Thanks for that. Example scripts weren't working I believe. I was going to try and install the psmoveapi from scratch separately then try to reinstall Joust but haven't had a chance yet. Thanks for the info, let me know how you go with your testing once you've had a chance.

anoopr, 2023-01-11:
I just went through this. Adding `git checkout 8a1f8d035e9c82c5c134d848d9fbb4dd37a34b58` on the line after `cd psmoveapi` at line 60 fixes this by using a version of psmoveapi from October. A change they've made since then breaks the Raspberry Pi build.

adangert, 2023-01-11:
Thanks for looking through this! I'll make the change and look into what changed on the psmove side of things

adangert, 2023-11-30:
From this commit here:
https://github.com/thp/psmoveapi/commit/1c9210e96bacd8d38febeb084ad4b3f7b46d3bb9
removed SWIG-based Python bindings
This will need to be updated for all files