adangert/JoustMania #290: Issues with installing Joustmania on latest Raspbian OS

supersmo, 2023-04-29:
I encountered several issues with installing Joustmania on the latest Raspbian OS image.

* User needs to be called pi because of `runuser -l pi` in update.py
* git is very picky about trusting folders and prompts about adding the repo directories to the safe listing in the global git config.
* It seems that pulseaudio does not work under root.
* On raspberry Pi 4 it complains about having pi3-disable-bt having been renamed to disable-bt. (but it still works, so it looks like a deprecation warning)

When the pi is rebooted I hear the menu music for half a second and then it goes silent.
I've been troubleshooting this for a while now but I'm giving up and going back to Buster to see if I can get it working.
We're having an event around May 4th (Star Wars day) and I have an idea of modding Joustmania to use Star Wars music, Yoda voice as presenter and light saber sound effects for joining. Hope it will work.

supersmo, 2023-04-30:
I tried with buster but encountered the same issues.
Output of the err log:
```
pi@raspberrypi:~ $ cat /var/log/joustmania-err.log
Home directory not accessible: Permission denied
Home directory not accessible: Permission denied
Home directory not accessible: Permission denied
/usr/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: Your system is neon capable but pygame was not built with support for it. The performance of some of your blits could be adversely affected
  return f(*args, **kwds)
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:80
 * Running on http://127.0.0.1:80
Press CTRL+C to quit
fatal: unable to access 'https://github.com/adangert/JoustMania.git/': Could not resolve host: github.com
fatal: detected dubious ownership in repository at '/home/pi/JoustMania'
To add an exception for this directory, call:

        git config --global --add safe.directory /home/pi/JoustMania
/usr/lib/python3/dist-packages/scipy/signal/signaltools.py:2223: FutureWarning: Using a non-tuple sequence for multidimensional indexing is deprecated; use `arr[tuple(seq)]` instead of `arr[seq]`. In the future this will be interpreted as an array index, `arr[np.array(seq)]`, which will result either in an error or a different result.
  Y[sl] = X[sl]
/usr/lib/python3/dist-packages/scipy/signal/signaltools.py:2225: FutureWarning: Using a non-tuple sequence for multidimensional indexing is deprecated; use `arr[tuple(seq)]` instead of `arr[seq]`. In the future this will be interpreted as an array index, `arr[np.array(seq)]`, which will result either in an error or a different result.
  Y[sl] = X[sl]
Process Process-4:
Traceback (most recent call last):
  File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/JoustMania/piaudio.py", line 178, in audio_loop
    WriteSamples(device, PERIOD_BYTES, Resample(ReadSamples(wf, PERIOD)))
  File "/home/pi/JoustMania/piaudio.py", line 154, in WriteSamples
    device.write(buf[:write_size])
alsaaudio.ALSAAudioError: Broken pipe [default]
Home directory not accessible: Permission denied
Home directory not accessible: Permission denied
Home directory not accessible: Permission denied
```

adangert, 2023-04-30:
Thanks for the info I'll look into this asap

supersmo, 2023-04-30:
Thanks.
`pulseaudio --version` shows `pulseaudio 12.2`

I found this but I'm not sure if it's what broke it:
```launch: avoid autostarting pulseaudio systemd units for root```
https://www.freedesktop.org/wiki/Software/PulseAudio/Notes/12.0/

I tried to see if there was an option to downgrade pulseaudio via apt but I didn't find a release to downgrade to.

RIght now pulseaudio is the biggest issue.

supersmo, 2023-04-30:
Pulse audio version 14 release notes:
```      Permit root-owned home directory```
https://www.freedesktop.org/wiki/Software/PulseAudio/Notes/14.0/

Might fix it???
But it needs to built from source and not sure if it's compatible with other components that are dependent on pulseaudio.

supersmo, 2023-04-30:
Noticed that pulse audio was already at 14.2 in bullseye in 2021-10-30, so there’s no point in compiling it from source for buster to try it out.

adangert, 2023-04-30:
Just pushed up a commit that seems to fix the issue:

previously a bunch of the settings for the device were being set in different lines:
device = alsaaudio.PCM()
device.setchannels(wf.getnchannels())
device.setformat(alsaaudio.PCM_FORMAT_S16_LE)
device.setperiodsize(PERIOD)

when combined to the default constructor:
device = alsaaudio.PCM(channels=wf.getnchannels(), rate=wf.getframerate(), format=alsaaudio.PCM_FORMAT_S16_LE, periodsize=PERIOD, device='default')

Then it seems to be fixed, it's probably something that alsa audio added in a recent updated. But this fix should be more stable going forward from now on I would imagine.

adangert, 2023-04-30:
I'll keep this issue open until the rest of the issues listed above are solved as well.

supersmo, 2023-04-30:
Thanks!
That’s great news! It’s around midnight here so I’ll test it tomorrow and report back after verifying the fix 👍

supersmo, 2023-05-01:
I started off with a fresh version of the latest Raspberry Pi Os, installed updates  and rebooted.
When installing Joustmania I now get an error:
```
Collecting pygame
  Downloading pygame-2.4.0.tar.gz (13.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 13.2/13.2 MB 6.8 MB/s eta 0:00:00
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'error'
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [30 lines of output]
      
      
      WARNING, No "Setup" File Exists, Running "buildconfig/config.py"
      Using UNIX configuration...
      
      /bin/sh: 1: sdl2-config: not found
      /bin/sh: 1: sdl2-config: not found
      /bin/sh: 1: sdl2-config: not found
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-25qwdjaf/pygame_a771dfc8954f45e5b22c8a4a753ba3a6/setup.py", line 400, in <module>
          buildconfig.config.main(AUTO_CONFIG)
        File "/tmp/pip-install-25qwdjaf/pygame_a771dfc8954f45e5b22c8a4a753ba3a6/buildconfig/config.py", line 231, in main
          deps = CFG.main(**kwds, auto_config=auto)
        File "/tmp/pip-install-25qwdjaf/pygame_a771dfc8954f45e5b22c8a4a753ba3a6/buildconfig/config_unix.py", line 218, in main
          DependencyProg('SDL', 'SDL_CONFIG', 'sdl2-config', '2.0', ['sdl']),
        File "/tmp/pip-install-25qwdjaf/pygame_a771dfc8954f45e5b22c8a4a753ba3a6/buildconfig/config_unix.py", line 43, in __init__
          self.ver = config[0].strip()
      IndexError: list index out of range
      
      Hunting dependencies...
      
      ---
      For help with compilation see:
          https://www.pygame.org/wiki/Compilation
      To contribute to pygame development see:
          https://www.pygame.org/contribute.html
      ---
      
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.
```
Looks like they pushed a new version of pygame and that's what broke it: https://pypi.org/project/pygame/#history
When I pinned it to 2.3.0 it worked. (Typing this on the pi so I'll post this before rebooting. Hopefully everything is working now.)

The issue with git needing the Joustmania folder to be added to safe directory when checking for updates to the Joustmania repo is because root owns the venv  folder and the setup.log file. I think `venv` and `setup.log` needs to be created as the normal user or `chown` needs to be run before the git commands.

dac1715, 2023-05-01:
I tried to install this game for the first time. I have no experience in any of this and just followed instructions. I have no audio.

I hope you guys fix it. Thanks in advance!

Regards!

supersmo, 2023-05-01:
Thanks @adangert it's working!
(Like mentioned earlier, I had to pin pygame in order for it to install properly)

@dac1715 What audio output are you listening to? I'm getting audio on the headphone socket and not on HDMI audio.

Did you get any error messages when you installed it with  `sudo ./setup.sh --disable_internal_bt`?

If you get something similar to what I posted you can modify the [setup.sh line 54](https://github.com/adangert/JoustMania/blob/master/setup.sh#L54)
Change it from:
```
$PYTHON -m pip install --ignore-installed psutil flask Flask-WTF pyalsaaudio pydub pygame pyaudio pyyaml dbus-python || exit -1
```
to:
```
$PYTHON -m pip install --ignore-installed psutil flask Flask-WTF pyalsaaudio pydub pygame==2.3.0 pyaudio pyyaml dbus-python || exit -1```

and then rerun the installation (sudo ./setup.sh...)

dac1715, 2023-05-01:
@supersmo I will try later with headphone socket, didn't have any speakers nearby (so HDMI is not working for me).

I don't know where to read error messages :D

I got notification "pi3-disable-bt having been renamed to disable-bt" but you mentioned that it still works.

I will ask a friend of mine to help me with setup.sh line change.

Ty!

supersmo, 2023-05-03:
@adangert Thanks for fixing so we could play!
We played 9 player JoustMania, modded with Star Wars music, sound effects, AI generated C-3PO speech explaining the rules for Werewolves mode, or “Sith among us” as it was renamed. Jedi lightsabers were blue and Sith lightsabers were red.
(Noticed that the setting to turn off red light when dying was broken so I did a quick dirty fix in the code.)
The two modes we played were “Joust random teams” and “Sith among us”.

We had a great time and they loved it! 👍
Thank you!

adangert, 2023-05-03:
🙌 I'm so glad it worked out! That theme change sounds so fun, and really on point with the different lightsaber colors and sounds!!

adangert, 2023-05-06:
Just finished updating all the files to fix the above errors. update.py, the pygame install and disable-bt problems have been fixed.

closed 2023-05-06 by adangert